ARMmbed/mbed-os #13500: STM32WB and mic MP34DT05-A with PDM - error when reading data in mbed

pawelm87, 2020-08-27:
### Description of defect

Hi all,

I try implement measuring sound level (SPL). I choose mic MP34DT05-A with PDM and connect to nucleo STM32WB55RG:
- data of mic connected to SAI_D1_PC3
- clock od mic connected to SAI1_CK1_PB8
- used the one mic so input of mic LR connected to GND (or VCC)

To generate code (setting of register) I used described example from section 4.3 of datasheet "Examples of configuration based on STM32CubeMX AN5027" and CubeMX. 
[image]
[image]
Then I got two functions for the comprehensive PDM configuration:
```
static void MX_SAI1_Init(void)
{

  /* USER CODE BEGIN SAI1_Init 0 */

  /* USER CODE END SAI1_Init 0 */

  /* USER CODE BEGIN SAI1_Init 1 */

  /* USER CODE END SAI1_Init 1 */
  hsai_BlockA1.Instance = SAI1_Block_A;
  hsai_BlockA1.Init.Protocol = SAI_FREE_PROTOCOL;
  hsai_BlockA1.Init.AudioMode = SAI_MODEMASTER_RX;
  hsai_BlockA1.Init.DataSize = SAI_DATASIZE_16;
  hsai_BlockA1.Init.FirstBit = SAI_FIRSTBIT_LSB;
  hsai_BlockA1.Init.ClockStrobing = SAI_CLOCKSTROBING_FALLINGEDGE;
  hsai_BlockA1.Init.Synchro = SAI_ASYNCHRONOUS;
  hsai_BlockA1.Init.OutputDrive = SAI_OUTPUTDRIVE_DISABLE;
  hsai_BlockA1.Init.NoDivider = SAI_MASTERDIVIDER_DISABLE;
  hsai_BlockA1.Init.MckOverSampling = SAI_MCK_OVERSAMPLING_DISABLE;
  hsai_BlockA1.Init.FIFOThreshold = SAI_FIFOTHRESHOLD_EMPTY;
  hsai_BlockA1.Init.MonoStereoMode = SAI_STEREOMODE;
  hsai_BlockA1.Init.CompandingMode = SAI_NOCOMPANDING;
  hsai_BlockA1.Init.PdmInit.Activation = ENABLE;
  hsai_BlockA1.Init.PdmInit.MicPairsNbr = 1;
  hsai_BlockA1.Init.PdmInit.ClockEnable = SAI_PDM_CLOCK1_ENABLE;
  hsai_BlockA1.FrameInit.FrameLength = 16;
  hsai_BlockA1.FrameInit.ActiveFrameLength = 1;
  hsai_BlockA1.FrameInit.FSDefinition = SAI_FS_STARTFRAME;
  hsai_BlockA1.FrameInit.FSPolarity = SAI_FS_ACTIVE_HIGH;
  hsai_BlockA1.FrameInit.FSOffset = SAI_FS_FIRSTBIT;
  hsai_BlockA1.SlotInit.FirstBitOffset = 0;
  hsai_BlockA1.SlotInit.SlotSize = SAI_SLOTSIZE_DATASIZE;
  hsai_BlockA1.SlotInit.SlotNumber = 1;
  hsai_BlockA1.SlotInit.SlotActive = 0x0000FFFF;
  if (HAL_SAI_Init(&hsai_BlockA1) != HAL_OK)
  {
    tr_info("error");
  }
  /* USER CODE BEGIN SAI1_Init 2 */
    tr_info("read status");
    tr_info("status sai (if'1' => sai ready to use): %d", HAL_SAI_GetState(&hsai_BlockA1));
    /* USER CODE END SAI1_Init 2 */
}

static void MX_DMA_Init(void) 
{

  /* DMA controller clock enable */
  __HAL_RCC_DMAMUX1_CLK_ENABLE();
  __HAL_RCC_DMA1_CLK_ENABLE();

  /* DMA interrupt init */
  /* DMA1_Channel1_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(DMA1_Channel1_IRQn, 0, 0);
  HAL_NVIC_EnableIRQ(DMA1_Channel1_IRQn);
}
```

So I put these funtions to simple mbed example code:

```
/* mbed Microcontroller Library
 * Copyright (c) 2018 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */

#include "mbed.h"
#include "ThisThread.h"
#include "mbed-trace/mbed_trace.h"

#define TRACE_GROUP  "MAIN"

DigitalOut led1(LED1);

SAI_HandleTypeDef hsai_BlockA1;

void Error_Handler(void)
{
    /* USER CODE BEGIN Error_Handler_Debug */
    /* User can add his own implementation to report the HAL error return state */
    tr_info("error");
    /* USER CODE END Error_Handler_Debug */
}

static void MX_SAI1_Init(void)
{

  /* USER CODE BEGIN SAI1_Init 0 */

  /* USER CODE END SAI1_Init 0 */

  /* USER CODE BEGIN SAI1_Init 1 */

  /* USER CODE END SAI1_Init 1 */
  hsai_BlockA1.Instance = SAI1_Block_A;
  hsai_BlockA1.Init.Protocol = SAI_FREE_PROTOCOL;
  hsai_BlockA1.Init.AudioMode = SAI_MODEMASTER_RX;
  hsai_BlockA1.Init.DataSize = SAI_DATASIZE_16;
  hsai_BlockA1.Init.FirstBit = SAI_FIRSTBIT_LSB;
  hsai_BlockA1.Init.ClockStrobing = SAI_CLOCKSTROBING_FALLINGEDGE;
  hsai_BlockA1.Init.Synchro = SAI_ASYNCHRONOUS;
  hsai_BlockA1.Init.OutputDrive = SAI_OUTPUTDRIVE_DISABLE;
  hsai_BlockA1.Init.NoDivider = SAI_MASTERDIVIDER_DISABLE;
  hsai_BlockA1.Init.MckOverSampling = SAI_MCK_OVERSAMPLING_DISABLE;
  hsai_BlockA1.Init.FIFOThreshold = SAI_FIFOTHRESHOLD_EMPTY;
  hsai_BlockA1.Init.MonoStereoMode = SAI_STEREOMODE;
  hsai_BlockA1.Init.CompandingMode = SAI_NOCOMPANDING;
  hsai_BlockA1.Init.PdmInit.Activation = ENABLE;
  hsai_BlockA1.Init.PdmInit.MicPairsNbr = 1;
  hsai_BlockA1.Init.PdmInit.ClockEnable = SAI_PDM_CLOCK1_ENABLE;
  hsai_BlockA1.FrameInit.FrameLength = 16;
  hsai_BlockA1.FrameInit.ActiveFrameLength = 1;
  hsai_BlockA1.FrameInit.FSDefinition = SAI_FS_STARTFRAME;
  hsai_BlockA1.FrameInit.FSPolarity = SAI_FS_ACTIVE_HIGH;
  hsai_BlockA1.FrameInit.FSOffset = SAI_FS_FIRSTBIT;
  hsai_BlockA1.SlotInit.FirstBitOffset = 0;
  hsai_BlockA1.SlotInit.SlotSize = SAI_SLOTSIZE_DATASIZE;
  hsai_BlockA1.SlotInit.SlotNumber = 1;
  hsai_BlockA1.SlotInit.SlotActive = 0x0000FFFF;
  if (HAL_SAI_Init(&hsai_BlockA1) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN SAI1_Init 2 */
    tr_info("read status");
    tr_info("status sai (if'1' => sai ready to use): %d", HAL_SAI_GetState(&hsai_BlockA1));
    /* USER CODE END SAI1_Init 2 */
}

static void MX_DMA_Init(void) 
{

  /* DMA controller clock enable */
  __HAL_RCC_DMAMUX1_CLK_ENABLE();
  __HAL_RCC_DMA1_CLK_ENABLE();

  /* DMA interrupt init */
  /* DMA1_Channel1_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(DMA1_Channel1_IRQn, 0, 0);
  HAL_NVIC_EnableIRQ(DMA1_Channel1_IRQn);

}

int main()
{    
    mbed_trace_init();
    mbed_trace_config_set(TRACE_ACTIVE_LEVEL_INFO);
    tr_info("START");

    MX_SAI1_Init();
    MX_DMA_Init();

    ThisThread::sleep_for(1000);

    uint16_t sound_len = 10;
    uint8_t sound_in[sound_len];
        
    while (true) {
        led1 = !led1;
        ThisThread::sleep_for(5000);
        uint8_t hal_status = HAL_SAI_Receive(&hsai_BlockA1, sound_in, 10, 1000);
        if (hal_status != HAL_OK)
        {
            tr_info("hal status (if '0' => OK): %d", hal_status);
            tr_info("status sai (if'1' => sai ready to use): %d,", HAL_SAI_GetState(&hsai_BlockA1));
            Error_Handler();
        } 
        else 
        {
            tr_info("hal status OK");
            tr_info("data: %s", mbed_trace_array(hsai_BlockA1.pBuffPtr, 10));
            tr_info("sound in %s", mbed_trace_array(sound_in, sound_len));
        }     
    }
}
```

To read mic data I used function:
`HAL_SAI_Receive(&hsai_BlockA1, sound_in, 10, 1000);`

but it recive error status HAL_ERROR

[image]

In mbed I don't change the clock setting so that may generate the error - I need connect clock to SAI periferia. Mbed doesn't give API to re-configure that - I don't want to do it on my own.
Or I made a configuration error.

someone can help me? I just need only read sound level not record or stream it anywhere.

#### Target(s) affected by this defect ?
STM32WB55RG

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 1.2.0

#### What version of Mbed-os are you using (tag or sha) ?

mbedOS 5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
[image]

jeromecoutant, 2020-08-27:
Hi
Maybe clock configuration is missing ?
( see SystemClock_Config in CubeMX generated main.c file)

pawelm87, 2020-08-27:
Hi I think so
```
/** Initializes the peripherals clocks 
  */
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_SMPS|RCC_PERIPHCLK_USART1
                              |RCC_PERIPHCLK_SAI1|RCC_PERIPHCLK_USB;
  PeriphClkInitStruct.PLLSAI1.PLLN = 24;
  PeriphClkInitStruct.PLLSAI1.PLLP = RCC_PLLP_DIV2;
  PeriphClkInitStruct.PLLSAI1.PLLQ = RCC_PLLQ_DIV2;
  PeriphClkInitStruct.PLLSAI1.PLLR = RCC_PLLR_DIV2;
  PeriphClkInitStruct.PLLSAI1.PLLSAI1ClockOut = RCC_PLLSAI1_SAI1CLK|RCC_PLLSAI1_USBCLK;
  PeriphClkInitStruct.Usart1ClockSelection = RCC_USART1CLKSOURCE_PCLK2;
  PeriphClkInitStruct.Sai1ClockSelection = RCC_SAI1CLKSOURCE_PLLSAI1;
  PeriphClkInitStruct.UsbClockSelection = RCC_USBCLKSOURCE_PLLSAI1;
  PeriphClkInitStruct.SmpsClockSelection = RCC_SMPSCLKSOURCE_HSI;
  PeriphClkInitStruct.SmpsDivSelection = RCC_SMPSCLKDIV_RANGE1;

  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
```

Can I overwrite the struct RCC_PeriphCLKInitTypeDef PeriphClkInitStruct and again re-init clock and the mbed don't crash ?

pawelm87, 2020-08-27:
@ciarmcom I would like add new feature, at the moment mbedOS don't support PDM

0xc0170, 2020-08-31:
> @ciarmcom I would like add new feature, at the moment mbedOS don't support PDM

@pawelm87 you should redirect this to forums rather (requesting a feature).

@jeromecoutant would this be better to continue in the forum (either mbed or cube related forum) ?

pea-pod, 2020-09-02:
Just a comment about where this should go, @0xc0170 and @jeromecoutant, I have had issues trying to get the PDM feature in both a Nucleo L452RE-P and one other board (cannot remember anymore).

I really wanted to use the built in DFSDM feature to interface to the X-Nucleo with the two PDM microphones built in. I tried to follow the ideas from [this repo](https://github.com/y2kblog/NUCLEO-L476RG_DFSDM_PDM-Mic), with the CubeMX software to generate the appropriate blocks and then shove all of that into an mbed based build. 

I was never successful. I never once received an interrupt for either a transaction or a DMA event.

I eventually gave up and found both an Adafruit I2S outputting mic that I happened to find at Micro Center and the I2S library from ST on the older mbed.com site. I was able to adapt it and get it working. Even that was some effort, but at least I could get DMA interrupts firing.

Perhaps it is better to have this in the forums, but this seems to me to be more of an issue if you are trying to combine the mbed OS and drivers with the DFSDM peripheral. Other people seem to not have problems with just using the CubeMX without also using mbed.

jeromecoutant, 2020-09-03:
Hi
I don't have any answer...
2 additional links in case you could find any tips:

- I2C on DMA implementation for STM32L4:
https://github.com/ABOSTM/mbed-os/commit/8b506787c35c697df7188c0c58529dde35eebbdb
Note this was only a proof of concept

- Audio demo code with STM32H7:
https://os.mbed.com/teams/ST/code/DISCO_H747I_AUDIO_demo/

...

SuperThunder, 2020-09-05:
@pea-pod are you able to post your modified I2S library? I am also trying to get I2S working with adafruit modules, and I'm running into compile errors even with a board relatively close to the supported ones (an F411CE).

pea-pod, 2020-09-05:
@SuperThunder I can probably do that. It is a modified version of the ST library, and so I'll make sure I understand the license as none of the files have it in them when I pulled from them.

@jeromecoutant: The repo comes from [https://os.mbed.com/teams/ST/code/ST_I2S/](https://os.mbed.com/teams/ST/code/ST_I2S/) but none of the files have licenses in the headers. I can put the Apache license that ST put in the os.mbed.com repo into a github one that I create, but I wanted to make sure that this was the right thing to do. It uses DMA, so there is a working example. That being said, I was hoping to use the DFSDM peripheral, and I could only get a different microphone (which includes its own DFSDM-like block inside) working that spit out I2S formatted data directly.

@SuperThunder: This works with a Nucleo F429ZI that I have. The original repo says it should work for a F401 and an F411. Looking at it, it should be modifiable without much more work to work with the WB series, as the code follows the mbed folder definition. But that being said, without having one in hand, I cannot help you too much with the debug.

One final comment: I did a lot of trial-and-error coding. I found out some details that were interesting to me which could be helpful to you. This is all in reference to the [Adafruit I2S MEMS Microphone Breakout - SPH0645LM4H](https://www.adafruit.com/product/3421), although I'm sure most mics which output I2S work similarly.
- The mic spits out signed integers.
- The mic transmits 32 bits per channel, two channels. If only one is connected, it should read all zeros. A bit high or low determines whether left or right channel is output first.
- The mic itself puts out 24 bits of actual data, with 18 of them meaningful. I believe the last 8 bits go high-Z, but that is beyond my memory at the moment.
- When the peripheral to memory transfer occurs, it does manage to unwind the bits and the half-words (a word being 32 bits for this processor). It does not correctly handle swapping the half-words. There is a processor instruction for swapping the two half-words. They are named something like `__REV` and can be found in the `mbed-os/cmsis/TARGET_CORTEX_M/` folder specific to your compiler (this is pre-restructuring -- I have not looked for it in the new one). This makes the byte and word swapping a single instruction, but you'll have to select the right one. You could allow the compiler to see the swap and handle it itself, but that sort of thing always makes me slightly anxious, when I really just want it to use that really nifty instruction.
- The DMA transfer setup in the F4 HAL expects _transfers_, where as the I2S driver follows the mbed convention of specifying _byte lengths_. This matters when trying to port over the I2S code to a specific device.
- The total bytes you need in an array is `word byte length * samples * spatial channel count`. Because the microphone transfers 18 bits of data, this means you need to use a signed 32 bit integer. If you want X samples, you will need X * 2 * 4 bytes in an array. 
- If your processing library can handle it, there's no good reason why you could not do a conversion where your stride is 4 bytes (or 8 bytes if you only care about left or right), but you're only converting two of the bytes to an `int16_t`. The only problem with this is that most libraries expect an array, and could get slower if they need to take a 32 bit number. I am not sure about this. I know that I managed to get the `arm math` library working, and in order to not prematurely optimize, I just ended up using a few different arrays for mental simplicity.
- I ended up using the F429ZI, because it has ethernet and I could spam my computer with unicast UDP packets to read samples (for use in spyder). The reason I avoided broadcast was that I literally brought my wireless network to a crawl sending to local broadcast. I do not suggest broadcast for this reason. (WiFi does broadcast packets very poorly, as far as I can tell.) I picked ethernet itself, because it was faster than serial, as I could not figure out how to send serial using DMA (or otherwise asynchronously) without dropping half-transfer lengths. I am not the best programmer, but I am not sure it was not actually caused by oddities in the asynchronous code inside the mbed driver. Your mileage may vary. I only have ST boards, so perhaps this is something specific to their implementation, but I do not think so. The drivers do funny things with the asynchronous portions.

I will try to post the library soon, preferably within a week.

jeromecoutant, 2020-09-07:
> @jeromecoutant: The repo comes from https://os.mbed.com/teams/ST/code/ST_I2S/ but none of the files have licenses in the headers. I can put the Apache license that ST put in the os.mbed.com repo into a github one that I create, but I wanted to make sure that this was the right thing to do.

Yes, thx!

SuperThunder, 2020-09-10:
@pea-pod Thank you for the details! I don't want to write too much more about I2S here since the original issue is for PDM, but thank you for sharing your experience and insights. I'm working with the INMP441 microphone and MAX98357 powered speaker, but I didn't get past compile errors (was something to do with missing functions in the F4 HAL) so specific hardware didn't matter. I look forward to reading through your modified library when you post it.

pea-pod, 2020-09-11:
> @pea-pod Thank you for the details! I don't want to write too much more about I2S here since the original issue is for PDM, but thank you for sharing your experience and insights.

No problem. It took me a long time to get to this point, and I gave up a number of paths I preferred. I wanted to stay with the STM32 and mbed, so I ended up with that I2S outputting mic.

> I'm working with the INMP441 microphone

I looked at the datasheet for this mic. The output is I2S, not a PDM based output. For the library, this is good, but I wanted to make sure that this is what you expected.

> but I didn't get past compile errors (was something to do with missing functions in the F4 HAL) so specific hardware didn't matter.

I was able to get my F429 to compile. I think if it's one of the three, it should work. Otherwise, it might be a datasheet exercise, but I do not think it is going to be that hard. I have been wrong before, but the library seems fairly straightforward. 

> I look forward to reading through your modified library when you post it.

I should do that soon. You should not be too impressed with _my_ contribution to it, since I didn't modify it that much, and it could possibly use some polish. The most impactful thing I did for my personal work was to figure out the byte ordering issue.

pea-pod, 2020-09-13:
@SuperThunder: Ok. I thought I would do more, but no. I have made the repo public at https://github.com/pea-pod/mbed-stm32-i2s

I wanted to add an example... but I did not. So let me know if you need one, but it should work similar to how the drivers work.

@jeromecoutant: Please let me know if this is sufficient for the license. I set the license date to 2017 as that's when I saw that the repo on mbed.com was initially released.

SuperThunder, 2020-09-14:
@pea-pod Thanks! I've downloaded it but ran into the same issue as I did with the original ST library. I posted an issue in your repository.

jeromecoutant, 2020-09-14:
@pawelm87 
About the original ticket issue, do you expect more help from community ?

0xc0170, 2020-09-14:
> About the original ticket issue, do you expect more help from community ?

would it make sense to close and continue there?

pawelm87, 2020-09-17:
Thanks to everyone for help. I closeing  it, but in the future may I need more info and reopen again topic.

closed 2020-09-17 by pawelm87