ARMmbed/mbed-os #13504: minimal-printf no output for Linux due to LONG_MAX usage

maxgerhardt, 2020-08-27:
### Description of defect

When compiling the minimal-printf code for a x64 Linux machine, by default the wrapped `printf()` function produces no output because of a wrong buffer size check regarding `INT_MAX` and `LONG_MAX`. The called function checks for `LONG_MAX`:

https://github.com/ARMmbed/mbed-os/blob/afcf91f3314d82ce8d555f4a891c9fd8db13910f/platform/source/minimal-printf/mbed_printf_implementation.c#L345-L353

However the wrapper function, when it wants to directly print to stdout with a unbuffered (or, unlimited buffer so to speak), it calls

https://github.com/ARMmbed/mbed-os/blob/afcf91f3314d82ce8d555f4a891c9fd8db13910f/platform/source/minimal-printf/mbed_printf_wrapper.c#L33-L41

The check inside the called function will alsways fail since the passed buffer size is beyond `INT_MAX`, and thus the function terminates without producing any output.

When replacing the calling code with

```cpp
    int result = mbed_minimal_formatted_string(NULL, INT_MAX, format,
                                               arguments, stdout);
```

output is produced as expected. 

#### Target(s) affected by this defect ?

Linux

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

afcf91f3314d82ce8d555f4a891c9fd8db13910f

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
GCC 9.3.0

#### How is this defect reproduced ? 

1. Download https://github.com/maxgerhardt/mbed-minimal-printf-linux
2. Compile and run the code in the fixed version, observer output
3. In `mbed_minimal_printf`, comment out the fixed version and use the original version as supplied in mbed-os
4. Recompile, run and observe *no output*

maxgerhardt, 2020-08-27:
May be because on ARM / many embedded devs `int = long = 32 bit` but on my system

```cpp
    mbed_minimal_printf("sizeof long = %d sizeof int = %d\n", (int)sizeof(long), (int)sizeof(int));
```

```
sizeof long = 8 sizeof int = 4
```
so incidently on ARM the `LONG_MAX` and `INT_MAX` is the same.

0xc0170, 2020-08-31:
cc @ARMmbed/mbed-os-core

0xc0170, 2020-08-31:
@maxgerhardt would you be able to fix this in minimal-printf ? Looks like you found the offender already.

maxgerhardt, 2020-09-01:
@0xc0170 done as linked above.