ARMmbed/mbed-os #13506: Impossible to use PF_13 ADC on STM32H743ZI

fabiencomte, 2020-08-28:
### Description of defect

Hello,
Using AnalogIn adc_AIR_PRESSURE_ADC(PF_13); makes a mbed_die call
I have another ADC that work. The specificity of this one is that it use ADC_2

#### Target(s) affected by this defect ?
STM32H743ZI

#### Toolchain(s) (name and version) displaying this defect ?
Mbed studio last revision

#### What version of Mbed-os are you using (tag or sha) ?
1c0318ccfe0dd18cba452577522eb5b6f526ea97

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed studio last revision

#### How is this defect reproduced ? 
Add in global AnalogIn adc_AIR_PRESSURE_ADC(PF_13); to a basic program and run it.

jeromecoutant, 2020-08-31:
Hi
Issue confirmed.
Analysis on going

fabiencomte, 2020-09-01:
Thank you jeromecoutant

PR #13523 (2020-09-03): STM32H7 ADC: No MultiMode configuration needed for ADC2
### Summary of changes 

Fixes #13506 

Multi mode configuration is needed only for ADC1.

@fabiencomte

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-09-03 by 0xc0170