ARMmbed/mbed-os #13510: Default USB Serial conflict with other Serials

masoudr, 2020-08-28:
### Description of defect

My target is a Nucleo F103RB board and I want to use two serial ports. One is a USB port and a USART1 serial.
I'm trying to change the default serial pins for printf as described in [here](https://os.mbed.com/teams/ST/wiki/STDIO) to use with USART1 instead of default USB and manually initialize USB serial in my code.

#### Target(s) affected by this defect ?
STM32F103RBT6

#### Toolchain(s) (name and version) displaying this defect ?
GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

`mbed_app.json`:
```
{
    "requires": ["bare-metal"],
    "macros": [
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "target_overrides": {
        "*": {
            "target.c_lib": "small",
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": false,
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```
Sample code:
```
#include "mbed.h"

static BufferedSerial USBSerial(USBTX, USBRX, 115200);

int main()
{
    const char* msg = "Message from USB Serial\r\n";
    USBSerial.write(msg, strlen(msg));
    // This should print to USART1
    printf("Message from USART1\r\n");
}
```
I get no results in the USB port and get both outputs in USART1:

> 22:49:13.109 -> Message from USB Serial
> 22:49:13.109 -> Message from USART1
>

masoudr, 2020-08-28:
@ciarmcom I've updated the issue. I think it is been completed now.

jeromecoutant, 2020-08-31:
Hi @masoudr 

From
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F1/TARGET_STM32F103xB/TARGET_NUCLEO_F103RB/PinNames.h#L126-L148

here is my understanding...

Default configuration:
    STDIO_UART_TX = PA_2,
    STDIO_UART_RX = PA_3,
    USBTX       = STDIO_UART_TX,
    USBRX       = STDIO_UART_RX,
So, USBSerial and printf mapped to UART2

With
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
    STDIO_UART_TX = MBED_CONF_TARGET_STDIO_UART_TX,
    STDIO_UART_RX = MBED_CONF_TARGET_STDIO_UART_RX,
    USBTX       = STDIO_UART_TX,
    USBRX       = STDIO_UART_RX,
So, USBSerial and printf mapped to UART1...

Could you try to change your code like this?
static BufferedSerial USBSerial(PA_2, PA_3, 115200);

masoudr, 2020-08-31:
Currently, I removed the pin configuration from `mbed_app.json` and created two buffers like this:
```
static BufferedSerial USBSerial(USBTX, USBRX, 115200);
static BufferedSerial DebugSerial(PA_9, PA_10, 115200);
```
Then I have separate data lines in both USB connection and UART1 pins and works fine without any conflict.
This chip has a separate USB connection that is connected to USB Pins (USBDM and USBDP equivalent to PA11 and PA12). I thought that in the Nucleo board they connected the USB connector directly to those pins.

jeromecoutant, 2020-08-31:
You can use USB pins after #13406 integration :-)

Correct mbed documentation page is
https://os.mbed.com/docs/mbed-os/v6.2/apis/usbserial.html

0xc0170, 2020-08-31:
@masoudr is this all resolved with following references or there is still an issue to be fixed?

masoudr, 2020-08-31:
@0xc0170 There was misunderstood with me about USBSerial but I'm still haven't convinced about the configuration because when I set these
```
{
      "target.stdio_uart_tx": "PA_9",
      "target.stdio_uart_rx": "PA_10"
}
```
It should set the default serial to UART1. And when I set `static BufferedSerial USBSerial(USBTX, USBRX, 115200);` I'm setting the UART2 serial. So as @jeromecoutant said if I change a serial port in `mbed_app.json` it would map `USBTX` and `USBRX` to those pins which I think is wrong.

jeromecoutant, 2020-09-09:
What I said is that issue is USBTX is defined as "target.stdio_uart_tx",
so whatever the stdio_uart_tx value you set in json, USBTX will follow...

Solution is to not use USBTX name, use USBSerial(PA_2, PA_3, 115200)

0xc0170, 2020-09-10:
I'll close as resolved

closed 2020-09-10 by 0xc0170