ARMmbed/mbed-os #13511: MBEDTLS_USER_CONFIG example in the Readme yields an error

mpg, 2020-08-30:
### Description of defect

I'm trying to use a "user config file" for Mbed TLS as describe in the Readme here: https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/features/mbedtls/README.md#configuring-mbed-tls-features

Here's my full `mbed_app.json` file:

```
{
    "macros" : [

        "MBEDTLS_USER_CONFIG_FILE" : "my_config.h",

        "MBEDTLS_PEM_WRITE_C",
        "MBEDTLS_CMAC_C"
    ]
}
```
With this file, `mbed compile` returns the following error:

```
Error parsing './mbed_app.json': Expecting ',' delimiter: line 4 column 36 (char 55)
[...]
[ERROR] Could not parse mbed app configuration from ./mbed_app.json
```

#### Target(s) affected by this defect ?

Probably all of them, but I was using `NUCLEO_F091RC`.

#### Toolchain(s) (name and version) displaying this defect ?

Probably all of them, but I was using `GCC_ARM`

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4 (commit hash: e4b81f67f939a0c0b11c147ce74aa367271e1279)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

`mbed --version`: `1.10.4`

#### How is this defect reproduced ? 

Use the `mbed_app.json` file above, verify that `mbed compile` fails, and that the `mbed_app.json` file was taken straight from the Readme that is part of Mbed OS.

I tried the obvious change in the `mbed_app.json` file:
```
{
    "macros" : {

        "MBEDTLS_USER_CONFIG_FILE" : "my_config.h",

        "MBEDTLS_PEM_WRITE_C": "",
        "MBEDTLS_CMAC_C": ""
    }
}
```
and it obviously failed with a different error:
```
[ERROR] in ./mbed_app.json element macros: OrderedDict([('MBEDTLS_USER_CONFIG_FILE', 'my_config.h'), ('MBEDTLS_PEM_WRITE_C', ''), ('MBEDTLS_CMAC_C', '')]) is not of type 'array'
```
which seems fair enough.

In the end I used `mbed compile '-DMBEDTLS_USER_CONFIG_FILE="my_config.h"' on the command line which worked as intended. If that's indeed the recommended way of using an Mbed TLS user config file, the Readme should say so.

0xc0170, 2020-08-31:
We shall fix the readme, `"MBEDTLS_USER_CONFIG_FILE=my_config.h"`

@ARMmbed/mbed-os-security Please review

mpg, 2020-09-01:
I didn't test, but I think `"MBEDTLS_USER_CONFIG_FILE=\"my_config.h\""` is more likely to work (there has to be quotes in the replacement text of the macro, or the preprocessor will complain about an unexpected token after `#include`).