ARMmbed/mbed-os #13519: Fail to compile Greentea test DIRECT_ACCESS_DEVICEKEY_TDB

MarceloSalazar, 2020-09-01:
### Description of defect

When running Greentea tests for Mbed Enabled validation, see a compile error with one specific test.

```
Build failures:
  * K64F::ARMC6::MBED-OS-STORAGE-KVSTORE-DIRECT_ACCESS_DEVICEKEY-TESTS-TESTS-DIRECT_ACCESS_DEVICEKEY-DIRECT_ACCESS_DEVICEKEY_TDB
        Building project direct_access_devicekey_tdb (K64F, ARMC6)
        Scan: ARM
        Scan: direct_access_devicekey_tdb
        Macros: -DTARGET_M4 -DTARGET_MCUXpresso_MCUS -DDEVICE_I2CSLAVE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DMBED_TEST_MODE -DTOOLCHAIN_ARM -DDEVICE_PORTIN=1 -DDEVICE_I2C=1 -DDEV
ICE_ANALOGIN=1 -DTARGET_MBED_PSA_SRV -DDEVICE_ANALOGOUT=1 -DTARGET_KSDK2_MCUS -DTARGET_KPSDK_CODE -D__CORTEX_M4 -DDEVICE_PWMOUT=1 -DTARGET_PSA_V7_M -DFSL_RTOS_MBED -DDEVICE_RESET_REASON=1 -
DTARGET_Freescale_EMAC -D__MBED_CMSIS_RTOS_CM -DARM_MATH_CM4 -DDEVICE_SLEEP=1 -DTOOLCHAIN_ARM_STD -D__CMSIS_RTOS -DDEVICE_USBDEVICE=1 -DDEVICE_SPISLAVE=1 -DDEVICE_PORTOUT=1 -DMBED_SPLIT_HEA
P -DDEVICE_SERIAL_FC=1 -DDEVICE_LPTICKER=1 -DDEVICE_SERIAL=1 -DFEATURE_PSA=1 -DTARGET_MCU_K64F -DTARGET_LIKE_CORTEX_M4 -DMBED_BUILD_TIMESTAMP=1598953389.7593205 -DTARGET_FRDM -D__FPU_PRESEN
T=1 -DDEVICE_SPI=1 -DTARGET_KPSDK_MCUS -DDEVICE_USTICKER=1 -DCOMPONENT_SD=1 -DDEVICE_PORTINOUT=1 -DTARGET_CORTEX_M -DTARGET_RELEASE -DDEVICE_CRC=1 -DTARGET_FF_ARDUINO -DTOOLCHAIN_ARMC6 -DCO
MPONENT_FLASHIAP=1 -D__MBED__=1 -DDEVICE_TRNG=1 -DDEVICE_FLASH=1 -DTARGET_LIKE_MBED -DDEVICE_WATCHDOG=1 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_PSA_Target -DTARGET_CORTEX -DDEVICE_EMAC=1 -DDEVICE
_RTC=1 -DMBED_TICKLESS -DTARGET_NAME=K64F -DTARGET_K64F -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_Freescale -DTARGET_RTOS_M4_M7
        Link: direct_access_devicekey_tdb
        Link: c:\Users\MS\AppData\Local\Mbed Studio\mbed-studio-tools\ac6\bin\armlink --ide=mbed --via .\BUILD\tests\K64F\ARM\.\mbed-os\storage\kvstore\direct_access_devicekey\tests\TE
STS\direct_access_devicekey\direct_access_devicekey_tdb\.link_options.txt
        [DEBUG] Return: 1
        [Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
        [DEBUG] Errors: Warning: L3912W: Option 'legacyalign' is deprecated.
        [DEBUG] Errors: Fatal error: L6002U: Could not open file BUILD/tests/K64F/ARM/mbed-os/storage/kvstore/direct_access_devicekey/tests/TESTS/direct_access_devicekey/direct_access_devic
ekey_tdb/mbed-os/storage/kvstore/direct_access_devicekey/tests/TESTS/direct_access_devicekey/direct_access_devicekey_tdb/main.o: No such file or directory
        [DEBUG] Errors: Not enough information to list the image map.
        [DEBUG] Errors: Finished: 1 information, 1 warning, 0 error and 1 fatal error messages.

[mbed] ERROR: "c:\Users\MS\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe" returned error.
       Code: 1
       Path: "c:\Users\MS\Mbed Programs\mbed-os-empty"
       Command: "c:\Users\MS\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe -u c:\Users\MS\Mbed Programs\mbed-os-empty\mbed-os\tools\test.py -D MBED_TEST_MODE -t AR
M -m K64F --source . --build .\BUILD\tests\K64F\ARM --test-spec .\BUILD\tests\K64F\ARM\test_spec.json --build-data .\BUILD\tests\K64F\ARM\build_data.json -n *DIRECT_ACCESS_DEVICEKEY_TDB* -v
 --greentea"
---
```

#### Target(s) affected by this defect ?

All (tested K64F, DISCO_L475VG_IOT01A)

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI (in Mbed Studio)

#### How is this defect reproduced ? 

mbed test -t GCC_ARM -m K64F --compile -n *DIRECT_ACCESS_DEVICEKEY_TDB* -v

0xc0170, 2020-09-14:
I recall this test was renamed just recently (@jeromecoutant  fixed it), can you retry on master

evedon, 2020-10-27:
```
~/git/tmp/mbed-os on master$ mbed test -t ARM -m K64F --compile -n *DIRECT_ACCESS_DEVICEKEY*
[mbed] Working path "/Users/evedon01/git/tmp/mbed-os" (program)
Building library mbed-build (K64F, ARMC6)
Scan: mbed-os
Building project tdb (K64F, ARMC6)
Scan: ARM
Scan: tdb
Link: tdb
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
Elf2Bin: tdb
| Module                   |     .text |   .data |      .bss |
|--------------------------|-----------|---------|-----------|
| [lib]/c_w.l              |  5210(+0) |  16(+0) |   348(+0) |
| [lib]/fz_wm.l            |    26(+0) |   0(+0) |     0(+0) |
| [lib]/libcppabi_w.l      |    44(+0) |   0(+0) |     0(+0) |
| anon$$obj.o              |    32(+0) |   0(+0) |  2304(+0) |
| cmsis/CMSIS_5            |  7356(+0) | 164(+0) |  1612(+0) |
| cmsis/device             |  2037(+0) |   4(+0) |  4997(+0) |
| connectivity/drivers     |     4(+0) |   0(+0) |     0(+0) |
| connectivity/nanostack   |   156(+0) |   0(+0) |     0(+0) |
| connectivity/netsocket   |   160(+0) |   0(+0) |     0(+0) |
| drivers/device_key       |   326(+0) |   0(+0) |     0(+0) |
| drivers/source           |  1452(+0) |   0(+0) |    80(+0) |
| features/frameworks      |  8248(+0) |  69(+0) |   433(+0) |
| hal/source               |  1724(+0) |   8(+0) |   131(+0) |
| platform/source          |  8409(+0) |  64(+0) |   349(+0) |
| rtos/source              |   342(+0) |   0(+0) |     0(+0) |
| storage/blockdevice      |  3486(+0) |   0(+0) |     0(+0) |
| storage/kvstore          | 11371(+0) |   0(+0) |   594(+0) |
| targets/TARGET_Freescale | 16980(+0) |  36(+0) |   494(+0) |
| Subtotals                | 67363(+0) | 361(+0) | 11342(+0) |
Total Static RAM memory (data + bss): 11703(+0) bytes
Total Flash memory (text + data): 67724(+0) bytes

Image: BUILD/tests/K64F/ARM/storage/kvstore/tests/TESTS/direct_access_devicekey/tdb/tdb.bin

Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|
| tdb  | K64F   | ARMC6     |      11703 |       67724 |

Build successes:
  * K64F::ARMC6::MBED-BUILD
  * K64F::ARMC6::STORAGE-KVSTORE-TESTS-TESTS-DIRECT_ACCESS_DEVICEKEY-TDB
```

The command line should be `mbed test -t ARM -m K64F --compile -n *DIRECT_ACCESS_DEVICEKEY*` and it works on master.

closed 2020-10-27 by evedon