ARMmbed/mbed-os #13528: SPIFBlockDevice: erase block calculation issue

boraozgen, 2020-09-02:
### Description of defect

I am using the [Winbond W25Q32JV SPI flash](https://www.winbond.com/resource-files/w25q32jv%20spi%20revc%2008302016.pdf) with the SPIFBlockDevice API. Here is the SPIF init debug log:

```
[83ms][DBG ][SFDP]: Detected Page Size: 256
[93ms][DBG ][SFDP]: Erase Type(A) 1 - Inst: 0xffffffffh, Size: 4096
[105ms][DBG ][SFDP]: Erase Type 1 - Inst: 0x20h, Size: 4096
[116ms][DBG ][SFDP]: Erase Type(A) 2 - Inst: 0xffffffffh, Size: 32768
[130ms][DBG ][SFDP]: Erase Type 2 - Inst: 0x52h, Size: 32768
[141ms][DBG ][SFDP]: Erase Type(A) 3 - Inst: 0xffffffffh, Size: 65536
[153ms][DBG ][SFDP]: Erase Type 3 - Inst: 0xd8h, Size: 65536
[164ms][DBG ][SFDP]: Erase Type(A) 4 - Inst: 0xffffffffh, Size: 1
[182ms][DBG ][SFDP]: Erase Type 4 - Inst: 0xffffffffh, Size: 1
[193ms][DBG ][SPIF]: Read Bus Mode set to 1-1-1, Instruction: 0x3h
[205ms][DBG ][SFDP]: No Sector Map Table
[216ms][DBG ][SPIF]: Device size: 4096 Kbytes
```

As seen on the log, it has 4KB, 32KB and 64KB erase blocks.

The issue arises when I try to erase a large block of the flash starting from the middle of a 32KB block. This causes the driver to send the command to erase the whole 32KB block, which also undesirably erases the data in the beginning of the block.

As an example, I call the erase method with the start address of 8192 (2 x 4KB blocks) and the size of 61440 (15 x 4KB blocks).

`storage.erase(8192, 61440);`

Ideally, this should erase the first 6 blocks using the 4KB erase command, the next 32KB block with the 32KB erase command and the last 4KB block with the corresponding command. However, it decides to erase the whole 32KB block where the the start address is in, which results in the erasure of the first 8192 bytes that I did not request. Here is the output:

```
[10007ms][DBG ][SPIF]: erase - addr: 8192, in_size: 61440
[10012ms][DBG ][SPIF]: erase - addr: 8192, size:61440, Inst: 0x52h, chunk: 24576 ,
[10019ms][DBG ][SPIF]: erase - Region: 0, Type:1
[10024ms][DBG ][SPIF]: Erase Inst: 0x52h, addr: 8192, size: 61440
[10159ms][DBG ][SPIF]: erase - addr: 32768, size:36864, Inst: 0x52h, chunk: 32768 ,
[10166ms][DBG ][SPIF]: erase - Region: 0, Type:1
[10171ms][DBG ][SPIF]: Erase Inst: 0x52h, addr: 32768, size: 36864
[10306ms][ERR ][SFDP]: No erase type was found for current region addr
[10312ms][DBG ][SPIF]: erase - addr: 65536, size:4096, Inst: 0x20h, chunk: 4096 ,
[10319ms][DBG ][SPIF]: erase - Region: 0, Type:0
[10324ms][DBG ][SPIF]: Erase Inst: 0x20h, addr: 65536, size: 4096
```

I believe that the erase block calculation algorithm should be updated to take this case into account.

#### Target(s) affected by this defect ?

Any target with an SPI external flash.

#### Toolchain(s) (name and version) displaying this defect ?

Not relevant.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant.

#### How is this defect reproduced ? 

Described above.

0xc0170, 2020-09-14:
> I believe that the erase block calculation algorithm should be updated to take this case into account.

Have you looked at the implementation, where is this calculation done ? And how to fix it?

boraozgen, 2020-09-15:
The calculation takes place here:

https://github.com/ARMmbed/mbed-os/blob/0db72d0cf26539016efbe38f80d6f2cb7a3d4414/storage/blockdevice/COMPONENT_SPIF/source/SPIFBlockDevice.cpp#L336-L379

Unfortunately I did not have the time to look into or fix the implementation.

LDong-Arm, 2020-11-23:
Hi @boraozgen, I've created a fix: #13947 
It would be great if you test it on your target and let us know, thanks!

PR #13947 (2020-11-26): Fix erase type determination for [Q/O/]BlockDevice::erase()

### Summary of changes 

Fixes: #13528 

`[Q/O/]BlockDevice::erase()` use `sfdp_iterate_next_largest_erase_type()` to determine the largest erase size suitable for the requested address and size, as it means better erase performance than smaller chunks. But there are several issues:

* `sfdp_iterate_next_largest_erase_type()`;
  * An alignment check is lacking. If the start address is aligned to a smaller erase unit but not a larger unit, this function wrongly returns the larger one, when the totally size to erase is larger than the  [... cut]
closed 2020-11-26 by 0xc0170