ARMmbed/mbed-os #13529: Cellular Connection Fails

bluePain, 2020-09-02:
### Description of defect

#### Target(s) affected by this defect ?
STM32F7-DISCO (I'm not sure if it's related)

#### Toolchain(s) (name and version) displaying this defect ?
Arm Gcc 6

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.0.0

#### How is this defect reproduced ? 
Connect Telit UL865 over UART on DISCO board and run Cellular Example.

0xc0170, 2020-09-14:
@ARMmbed/team-st-mcd would you be able to help/test ?

jeromecoutant, 2020-09-14:
Hi
How do you configure your mbed_app.json file ?
How the DISCO board is configured to use UART pins ?

jeromecoutant, 2020-10-01:
@bluePain any info?
Thx

bluePain, 2020-10-02:
@jeromecoutant Upss, sorry for skipping your comment.

I, unfortunately, disassembled the circuit and formatted my pc, but I'll have everything back again by the middle of the next week and then I'll post it.

bluePain, 2020-10-06:
I love that when it hits me with the cuty pinky bullet.

**[ERROR] armclang: error: System clock tampering detected. License checkout will fail.**

I was working on the MacOS when I post the issue, but now working on the Linux and cannot compile the code because of the licensing bug of arm-clang which is also stated in the [Known issues and troubleshooting](https://os.mbed.com/docs/mbed-studio/current/known-issues/index.html)

_§2 - Some users have reported licensing issues for Arm Compiler 6. These issues may be related to the system clock and region format._

I lost the last configuration while replacing my laptop and now trying to re-compile the code to be sure about what configuration I'm using when compiling. I'll post again as soon as fixing this issue.

bluePain, 2020-11-15:
Hi there,

I bricked the eval kit unfortunately. 

Since I may not be able to reproduce again, we can close it if you want until I get another one and start working with the same Teli chip.

bluePain, 2021-01-04:
I don't believe if I'll ever be able to re-create the same development environment and reproduce the error. 

Closing the issue to decrease the number of open (and stalled) issues.

closed 2021-01-04 by bluePain