ARMmbed/mbed-os #13532: Green Tea Tests Failing on K64F.

ThePhilomathsApprentice, 2020-09-02:
### Description of defect
I tried to run green tea tests for netsocket-tcp, but they are failing with hard fault and eventually timeout.

I have attached the very verbose log for the same.
please let me know if I need to add anything else.
Thanks.

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
toolchain: GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6-2-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli: 1.10.4
Greentea test automation tool ver. 1.7.4

#### How is this defect reproduced ? 
Step_1) mbed new mbedTest
Step_2) cd mbedTest
Step_3)  mbed test -m K64F -t GCC_ARM -n mbed-os-connectivity-netsocket-tests-tests-netsocket-tcp -vv > ../Test_Logs/TCP_Testv6_2020_09_02_1400.txt
[TCP_Testv6_2020_09_02_1400.txt](https://github.com/ARMmbed/mbed-os/files/5162612/TCP_Testv6_2020_09_02_1400.txt)

ThePhilomathsApprentice, 2020-09-02:
Can you please describe how to extract this sha or valid tag? Because I can't find it by using *mbed releases*.

Thanks.

0xc0170, 2020-09-03:
Ive checked the latest nightly results for K64F:

```
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-netsocket-dns                                                        | OK     | 44.73              | default     |
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-netsocket-tcp                                                        | OK     | 56.77              | default     |
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-netsocket-tls                                                        | OK     | 78.74              | default     |
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-netsocket-udp                                                        | OK     | 45.32              | default     |
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-network-interface                                                    | OK     | 48.82              | default     |
| K64F-GCC_ARM | K64F          | connectivity-netsocket-tests-tests-network-l3ip                                                         | OK     | 18.65              | default     |
```

You can check the logs here: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/619/greentea-test/261/PASS/K64F/GCC_ARM/

ThePhilomathsApprentice, 2020-09-03:
Hi @0xc0170 , I tested it again, It got passed, however I forgot to add one small step ( STEP_6 ) which I had done earlier but forgot to mention. I believe Steps 3 and 4 won't matter much as I'm already passing those arguments while testing.
Step_1) mbed new mbedTest
Step_2) cd mbedTest

Step_3) mbed target K64F
Step_4) mbed toolchain GCC_ARM

Step_5) mbed test -m K64F -t GCC_ARM -n mbed-os-tests-netsocket-tcp >> ../Test_Logs/TCP_Testv6_2020_09_03_2130_Screenlog.txt ( TEST_PASSED )

Step_6) mbed export -i mcuxpresso

step_7) mbed test -m K64F -t GCC_ARM -n mbed-os-tests-netsocket-tcp -vv >> ../Test_Logs/TCP_Testv6_2020_09_03_2145_Screenlog.txt ( test case results: 4 SKIPPED / 18 OK / 1 ERROR )

ps: I have also attached both the files for your reference.

Thanks,
Yash.

[TCP_Testv6_2020_09_03_2130_Screenlog.txt](https://github.com/ARMmbed/mbed-os/files/5170395/TCP_Testv6_2020_09_03_2130_Screenlog.txt)

[TCP_Testv6_2020_09_03_2145_Screenlog.txt](https://github.com/ARMmbed/mbed-os/files/5170396/TCP_Testv6_2020_09_03_2145_Screenlog.txt)

ThePhilomathsApprentice, 2020-09-07:
Hi @0xc0170 , Can you please tell me what command did you use for getting the above log file, It has the number of sockets opened in the tcp test, I am not getting that part in my TCP test log.

Thanks.

0xc0170, 2020-09-14:
Is there only difference in step 5 and 7 ` -vv` - verbose mode?

> Step_6) mbed export -i mcuxpresso

This step is not needed at all, does it make a difference for the next step (7) ?

ThePhilomathsApprentice, 2020-09-14:
1) Yes the difference is only -vv. It wasn't intentional. I added it only to see the detailed log.

2) Yes, this step is making a difference for me. After exporting it to mcuxpresso, the tests weren't passing.

0xc0170, 2020-09-15:
Thanks for confirming. As tools are frozen (new tools will be released), we can keep this opened (open for community contributions).