ARMmbed/mbed-os #13535: ADC errors on STM32H743 (wrong PLL config?)

isaev-d, 2020-09-02:
System error occurs while initializing AnalogIn

```
#include <mbed.h>
#include "rtos.h"

#define MBED_CONF_MBED_TRACE_ENABLE 1
#include "mbed-trace/mbed_trace.h"
#define TRACE_GROUP  "main"

int main() {  
    mbed_trace_init(); 

    AnalogIn   stick(PA_3);
    DigitalOut led(PA_1);

    while (1)
    {
        led = !led;
        tr_info("normalized: %04d\r\n", stick.read_u16());
        ThisThread::sleep_for(100ms);
    }
}
```
JTAG step-by-step debugging shows that error produced by HAL code :

analogin_device.c line 118
```
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
        error("analogin_init HAL_RCCEx_PeriphCLKConfig");
    }
```

steps: 
stm32h7xx_hal_rcc_ex.c line 3233
```
    /* Configure PLL2 multiplication and division factors. */
    __HAL_RCC_PLL2_CONFIG(pll2->PLL2M,
                          pll2->PLL2N,
                          pll2->PLL2P,
                          pll2->PLL2Q,
                          pll2->PLL2R);
```
sets unsupported M and N, 4 and 240, resulting PLL can't start in any timeout - PLL freq is out of range. STM32CubeMX shows default M and N 32 and 129.

In case of PLL settings corrected in analogin_device.c lines 109 and 110 to default from cube, error wont reproduce. 

However stick.read_u16() works only once, and then next error  occurs:
```
[INFO][main]: normalized: 14887

[INFO][main]: normalized: 0

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x80082C7
Error Value: 0x0
Current Thread: main Id: 0x24000CE8 Entry: 0x8002781 StackSize: 0x1000 StackMem: 0x240014E8 SP: 0x24002398 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_H743ZI2
-- MbedOS Error Info --
HAL_ADC_ConfigChannel issue
```

6.2.0

Board - DevEBox STM32H7XX (H743VI). Build with PlatformIO + VS Code on Linux system. On ini board selected as Nucleo H743. I know chip part number differ, but it is just wiring in another package.

```
Processing nucleo_h743zi (platform: ststm32; board: nucleo_h743zi; framework: mbed)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_h743zi.html
PLATFORM: ST STM32 8.0.0 > ST Nucleo H743ZI
HARDWARE: STM32H743ZIT6 400MHz, 512KB RAM, 2MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-mbed 6.60200.200722 (6.2.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
```

jeromecoutant, 2020-09-03:
Hi

In your PeripheralPins.c file, is PA_3 connected to ADC1 or ADC2 ?

FYI: there was an issue with ADC2 corrected with #13523

isaev-d, 2020-09-03:
{PA_3,       ADC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 15, 0)}, // ADC1_INP15

isaev-d, 2020-09-04:
Hm, filling myself an idiot..

Just tried sample code from  [os.mbed.com/docs/mbed-os/v6.2/apis/i-o-apis](https://os.mbed.com/docs/mbed-os/v6.2/apis/i-o-apis.html)

```
include "mbed.h"

AnalogIn input(PA_3);

#define NUM_SAMPLES 1024

int main()
{
    uint16_t samples[NUM_SAMPLES];

    for (int i = 0; i < NUM_SAMPLES; i++) {
        samples[i] = input.read_u16();
        ThisThread::sleep_for(1);
    }

    printf("Results:\n");
    for (int i = 0; i < NUM_SAMPLES; i++) {
        printf("%d, 0x%04X\n", i, samples[i]);
    }
}
```

And.. it works. But if sleep timeout is more than 4 ms, second conversion is corrupted, and next conversion produce error. Looks like ADC configured to series of conversions, not to single.

jeromecoutant, 2020-09-07:
Hi
I confirm the issue with series of conversions...

jeromecoutant, 2020-09-08:
@isaev-d could you check #13572 ?
Thx

closed 2020-09-09 by 0xc0170

isaev-d, 2020-09-09:
Yap, now it works fine. Many thanks, @jeromecoutant !

isaev-d, 2020-09-10:
I forgot to say that patch works only in case of right PLL configuration. In commit https://github.com/ARMmbed/mbed-os/pull/13572/commits that configuration is wrong. I'm just broke my workspace and after updating get PLL config error, look at top of post.

I think I would create another issue, as clock configuration on H743 platform should be fixed globally.

PR #13572 (2020-09-09): STM32H7 ADC: clock selection lost after deepsleep
### Summary of changes 

This fixes #13535 

PLL configuration was lost  during deepsleep.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type