ARMmbed/mbed-os #13539: BLE TARGET_CORDIO does not compile with MBED_NO_GLOBAL_USING_DIRECTIVE aktivated

EmbedEngineer, 2020-09-03:
### Description of defect
When compiling with MBED_NO_GLOBAL_USING_DIRECTIVE flag activated and using the Cordio stack, there are compiletime errors since there are no namespaces used in 
`mbed-os/connectivity/FEATURE_BLE/libraries/TARGET_CORDIO/driver/H4TransportDriver.h`.
E.g.:
`'UnbufferedSerial' does not name a type`, `'SerialBase' has not been declared`, etc.
There may be more files involved but this is the first one this error pops up.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Just compile with MBED_NO_GLOBAL_USING_DIRECTIVE added in mbed_app.json macros section.

0xc0170, 2020-09-03:
Thanks @EmbedEngineer . would you be able to send a fix addressing this?

cc @ARMmbed/mbed-os-pan

EmbedEngineer, 2020-09-04:
> Thanks @EmbedEngineer . would you be able to send a fix addressing this?
> 
> cc @ARMmbed/mbed-os-pan
Yeah I can try and fix it, but it will take some time since I am quite busy at the moment.