ARMmbed/mbed-os #13556: Memory Leak: Deleting a device Cellular_Context does not delete heap allocated Cellular_Stack

bsparker, 2020-09-06:
When deleting a dynamically allocated Cellular_Context (eg. `QUECTEL_BG96_CellularContext`), the destructor neglects deleting the CellularStack allocated by the `::get_stack()` function (eg. `QUECTEL_BG96_CellularContext::get_stack()`).

This is a memory leak.

#### Target(s) affected by this defect ?

A quick scan of the source files implies that none of the device CellularContext implementations delete the allocated Stack.

#### Toolchain(s) (name and version) displaying this defect ?

GNU Arm Embedded Toolchain 9-2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4 cellular core

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Custom cmake-based build

#### How is this defect reproduced ? 

Dynamically construct/use a cellular device (eg. QUECTEL_BG96), then delete. The destructor for the CellularStack is never called, because the CellularContext does not delete it.

bsparker, 2020-09-08:
@0xc0170 I'm not sure if its best/correct to delete it in the `CellularContext` base class destructor, or if it needs to be done in the child device classes (eg. `QUECTEL_BG96_CellularContext`). `CellularContext` has a pointer to a `NetworkStack`, and I don't think the `CellularContext` could know in all cases whether or not the `NetworkStack` pointer was heap allocated.

0xc0170, 2020-09-14:
> @bsparker it has been 5 days since the last reminder. Could you please update the issue header as previously requested?

Updated, description header added.

chrisswinchatt-arm, 2021-04-06:
Hi @b-scramps,

It looks like this was already fixed here at dd73a93

closed 2021-04-06 by pan-