ARMmbed/mbed-os #13577: nRF52 does not support baud rate 31250 for MIDI

gutbag, 2020-09-08:
The MIDI baud rate 31250 is not supported for Nordic's nRF52. The serial_baud() function in serial_api.c checks the requested baud rate against the list of supported values and uses the appropriate NRF_UARTE_BAUDRATE_xxx value, but 31250 is missing. But nrf_uarte.h defines NRF_UARTE_BAUDRATE_31250, so it could be added easily (I've done it locally, and I'm happy to issue a PR).

#### Target(s) affected by this defect ?

Nordic nRF52

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM but presumably this is N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.2.1 (1.2.1.4)

#### How is this defect reproduced ? 

Configure an instance of BufferedSerial with the baud rate 31250 and write some values. A receiver set to the correct baud rate will show incorrect values (if any).

0xc0170, 2020-09-10:
> The MIDI baud rate 31250 is not supported for Nordic's nRF52. The serial_baud() function in serial_api.c checks the requested baud rate against the list of supported values and uses the appropriate [...]

@gutbag if you tested it and works, please send a new pull request. Thanks

0xc0170, 2020-09-14:
Thanks for #13595, this will be fixed soon.

PR #13595 (2020-10-16): Nordic: Support 31250 baud rate
### Summary of changes

Add support for MIDI baud rate 31250 on Nordic nRF52 targets. Fixes #13577 

#### Impact of changes

#### Migration actions required

### Documentation

None

----------------------------------------------------------------------------------------------------------------
### Pull request type

    [X] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Return code change / API behaviour change)

-------- [... cut]
closed 2020-10-16 by 0xc0170