ARMmbed/mbed-os #13581: BLE: Intrinsically created descriptors (ie: CCCD) not accessible thru GattCharacteristic::getDescriptor()

AGlass0fMilk, 2020-09-09:
### Description of defect

Descriptors that are created intrinsically in the BLE stack are not accessible through `GattCharacteristic::getDescriptor()` as is semantically implied.

The only example of an intrinsically-created descriptor scenario I can think of is a `GattCharacteristic` that has `notify` and/or `indicate` properties enabled. The BLE stack will automatically create a `CCCD` descriptor attached to this `GattCharacteristic` so a `GattClient` can enable/disable notifications/indications.

I was trying to check if a certain characteristic's CCCD was written to in a callback handler that gets executed whenever any write to the GattServer happens. To do this, I need the CCCD handle of that characteristic. Since this is not available through `GattCharacteristic::getDescriptor()` I am currently just adding 1 to the handle of the `GattCharacteristic` itself. This is _typically_ the CCCD handle but this is not guaranteed anywhere in the BLE specification and so is potentially unreliable.

#### Target(s) affected by this defect ?

All BLE targets

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

3b5ab54618733e7d546410059d5f116b6477595c

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 

See above description.

AGlass0fMilk, 2020-09-16:
@pan- @paul-szczepanek-arm

pan-, 2020-09-17:
@AGlass0fMilk We're aware the API is not consistent at the moment, I can suggest several workaround: 
- Add a CCCD in the list of descriptors, when the characteristic is registered, the CCCD in the list will be used and the stack won't create one. As a consequence, the handle of the CCCD will be visible to the application. 
- Use the functions `GattServer::onUpdatesEnabled/onUpdatesDisabled` to detect when updates are enabled/disabled.

AGlass0fMilk, 2020-10-06:
> @AGlass0fMilk We're aware the API is not consistent at the moment, I can suggest several workaround:
>
>     * Add a CCCD in the list of descriptors, when the characteristic is registered, the CCCD in [...]

I am currently using the `GattServer::onUpdatesEnabled/onUpdatesDisabled` callbacks. The parameter passed by the stack into this callback is a `GattAttribute::Handle_t`. Similar to writing an `onDataWritten` handler, the `onUpdatesEnabled/onUpdatesDisabled` handler has to first check if this `GattAttribute::Handle_t` is relevant (ie: it is related to a characteristic within the service subscribing to the updates callback).

Is there any plan to revamp this API in the near future?

pan-, 2020-10-06:
I'm afraid there is no plan to refresh the Service construction API in the near future (PR are welcomed though!). 
In the short term it should be possible to add a function to `GattCharacteristic` to retrieve the CCCD handle.

AGlass0fMilk, 2020-10-06:
> I'm afraid there is no plan to refresh the Service construction API in the near future (PR are welcomed though!).
> In the short term it should be possible to add a function to `GattCharacteristic` to retrieve the CCCD handle.

Alright, I'll look into this. I just wanted to make sure I wasn't going to be duplicating any work if you guys were already revamping this API anyway.

PR #13729 (2020-10-13): BLE: Enable getting an implicitly-created CCCD through `GattCharacteristic::getDescriptor`

### Summary of changes 

See #13581 for information on issue.

Resolves #13581 by allowing the GattServer to pass a pointer to an implicitly-created CCCD. This `GattAttribute` is dynamically allocated (using `new`) and ownership is transferred to the relevant `GattCharacteristic`, which takes care of `delete`ing the `GattAttribute` in its destructor.

~I tried to make `GattServer` a `friend` class of `GattCharacteristic` so `GattCharacteristic::_setImplicitCCCD` could be `private`, but I wasn't able to  [... cut]
closed 2020-10-13 by 0xc0170