ARMmbed/mbed-os #13596: PSA force generate ROT

moshe-shahar, 2020-09-10:
### Description of defect

Applications that use PSA are forced to call `generate_root_of_trust` before using PSA after https://github.com/ARMmbed/mbed-os/pull/12385 was merged because current PSA implementation use kvstore to emulate storage.

#### Target(s) affected by this defect ?
PSA targets

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.1
master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced? 
Any test or application that uses PSA

moshe-shahar, 2020-09-10:
PR with fix https://github.com/ARMmbed/mbed-os/pull/13594

0xc0170, 2020-09-14:
> PR with fix #13594

Thanks, fixed now on master.

closed 2020-09-14 by 0xc0170