ARMmbed/mbed-os #13598: 32K File size limit passed to GCC_ARM compiler through @ flag

mbedNoobNinja, 2020-09-11:
### Description of defect

When the file **`.includes_xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.txt`** becomes larger than **32KB**,
the GCC compiler refuses to compile anything.

**Cmd**:

> arm-none-eabi-gcc -mcpu=cortex-a9 -march=armv7-a -marm -mthumb-interwork -mfloat-abi=hard -mfpu=vfpv3 -mno-unaligned-access -Os -fmessage-length=0 -ffunction-sections -fdata-sections -funsigned-char -fno-delete-null-pointer-checks -fomit-frame-pointer -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers  -g -DTARGET_RZA1H -D__FPU_PRESENT -DDEVICE_EMAC=1 -D__MBED__=1 -DDEVICE_USBDEVICE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -DDEVICE_PORTIN=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_VK_RZ_A1H -D__CMSIS_RTOS -DDEVICE_ETHERNET=1 -DCOMPONENT_SD=1 -D__EVAL -DTOOLCHAIN_GCC -DTARGET_CORTEX_A -DDEVICE_CAN=1 -DARM_MATH_CA9 -DDEVICE_I2C_ASYNCH=1 -DCOMPONENT_PSA_SRV_EMUL=1 -D__MBED_CMSIS_RTOS_CA9 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_LIKE_CORTEX_A9 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DTARGET_RZ_A1_EMAC -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DMBED_BUILD_TIMESTAMP=1595405206.25 -DDEVICE_USTICKER=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_RENESAS -DTARGET_RELEASE -DCOMPONENT_NSPE=1 -DDEVICE_SERIAL_FC=1 -DTARGET_A9 -D__CORTEX_A9 -DDEVICE_I2CSLAVE=1 -DTARGET_RZ_A1XX -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_VKRZA1H -DDEVICE_SPI=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_NAME=VK_RZ_A1H -include"D:/mbed_DIO/E2_IDE/mbed-src_gcc/mbed_config.h" -std=gnu11 -fno-exceptions -DMBED_TRAP_ERRORS_ENABLED=1 -DAPPLICATION_ADDR=0x18020000 -DAPPLICATION_SIZE=0x1fe0000 -DMBED_ROM_START=0x18020000 -DMBED_ROM_SIZE=0x1fe0000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0xa00000 -MMD -MF"mbed-os/targets/TARGET_RENESAS/TARGET_RZ_A1XX/common/rza_io_regrw.d" -MD @"./.includes_xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.txt" -c -o "mbed-os/targets/TARGET_RENESAS/TARGET_RZ_A1XX/common/rza_io_regrw.o" "D:/mbed_DIO/mbed-src/mbed-os/targets/TARGET_RENESAS/TARGET_RZ_A1XX/common/rza_io_regrw.c"

**Error**:

> arm-none-eabi-gcc: fatal error: cannot execute 'c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/as.exe': CreateProcess: No such file or directory
> compilation terminated.

#### Target(s) affected by this defect ?

`VK_RZ_A1H`

#### Toolchain(s) (name and version) displaying this defect ?

`GCC_ARM` 9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-5.15.4`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

| Tool | Version |
| --- | --- |
| appdirs | 1.4.3 |
| asn1ate | 0.6.0 |
| asn1crypto | 0.24.0 |
| backports.functools-lru-cache | 1.6.1 |
| beautifulsoup4 | 4.6.0 |
| cbor | 1.0.0 |
| certifi | 2018.1.18 |
| cffi | 1.11.5 |
| chardet | 3.0.4 |
| Click | 7.0 |
| cmsis-pack-manager | 0.2.4 |
| colorama | 0.3.9 |
| cryptography | 2.9.2 |
| ecdsa | 0.13 |
| enum34 | 1.1.6 |
| fasteners | 0.14.1 |
| funcsigs | 1.0.2 |
| functools32 | 3.2.3.post2 |
| future | 0.16.0 |
| futures | 3.2.0 |
| fuzzywuzzy | 0.16.0 |
| hidapi | 0.7.99.post21 |
| icetea | 1.2.1 |
| idna | 2.6 |
| intelhex | 2.2.1 |
| intervaltree | 2.1.0 |
| ipaddress | 1.0.22 |
| Jinja2 | 2.10.3 |
| jsonmerge | 1.5.2 |
| jsonschema | 2.6.0 |
| junit-xml | 1.8 |
| lockfile | 0.12.2 |
| Logbook | 1.4.1 |
| lxml | 4.2.5 |
| manifest-tool | 1.5.2 |
| MarkupSafe | 1.0 |
| mbed-cli | 1.10.4 |
| mbed-cloud-sdk | 2.2.0 |
| mbed-flasher | 0.10.1 |
| mbed-greentea | 1.7.4 |
| mbed-host-tests | 1.5.10 |
| mbed-ls | 1.7.12 |
| mbed-os-tools | 0.0.15 |
| milksnake | 0.1.5 |
| mock | 2.0.0 |
| monotonic | 1.4 |
| pbr | 3.1.1 |
| pip | 20.1.1 |
| prettytable | 0.7.2 |
| protobuf | 3.5.2.post1 |
| psutil | 5.6.6 |
| py | 1.7.0 |
| pyasn1 | 0.2.3 |
| pycparser | 2.19 |
| pycryptodome | 3.9.8 |
| pyelftools | 0.24 |
| pyocd | 0.14.3 |
| pyOpenSSL | 18.0.0 |
| pyparsing | 2.3.0 |
| pyserial | 3.4 |
| pyshark-legacy | 0.3.8 |
| python-dateutil | 2.7.5 |
| python-dotenv | 0.9.1 |
| pytz | 2020.1 |
| pyusb | 1.0.2 |
| pywin32 | 224 |
| pywinusb | 0.4.2 |
| PyYAML | 4.2b1 |
| requests | 2.20.1 |
| semver | 2.8.1 |
| setuptools | 28.8.0 |
| six | 1.12.0 |
| sortedcontainers | 2.1.0 |
| soupsieve | 1.9.6 |
| trollius | 1.0.4 |
| urllib3 | 1.24.2 |
| websocket-client | 0.46.0 |
| WMI | 1.4.9 |
| yattag | 1.10.1 |

#### How is this defect reproduced ? 

Find any compilable code, structured in as much folders as it possible, place it in the root directory of the mbed project (for me that was the `mbed-gr-libs` + `custom libs`  for streaming a video over LAN) and watch the .includes_x...x.txt file prodused by the python scripts. When the file grows beyond 32K (let's say 33) the error will happen for sure.

jeromecoutant, 2020-09-11:
Hi

I confirm I also sometimes got this error line
(arm-none-eabi-gcc: fatal error: cannot execute 'c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/as.exe': )

I am working on master branch

0xc0170, 2020-09-14:
Thanks for the report. This is known issue with the current tools (path limit in windows). Even response files are no help here.

The new upcoming work with CMake and components should address this (dont expose everything everywhere).

I'll close this as "won't fix", as there is not possible to fix this for the current tools (they were frozen). Thanks for understanding.

closed 2020-09-14 by 0xc0170

jeromecoutant, 2020-09-18:
> This is known issue with the current tools (path limit in windows).

I didn't understand why this issue came now ?

jeromecoutant, 2020-09-18:
For information, my current workaround is to create a .mbedignore file with:
````
mbed-os/connectivity/nanostack*
mbed-os/connectivity/drivers/nfc*
mbed-os/connectivity/nfc*
````