ARMmbed/mbed-os #13599: printf 、sprintf   %03d  format output  error

fengshengchang, 2020-09-11:
### Description of defect

I ran the following code, but I couldn't get Correct result. Result is "data:2", but Correct result should be "data:002"

int main()
{
   printf("data:%03d \r\n", 2);
}

#### Target(s) affected by this defect ?
NUCLEO_F103RB

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-cpp (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)

#### What version of Mbed-os are you using (tag or sha) ?

   mbed-os-6.2.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Package            Version
------------------ -----------
appdirs            1.4.4
asn1ate            0.6.0
attrs              20.2.0
beautifulsoup4     4.9.1
cbor               1.0.0
certifi            2020.6.20
cffi               1.14.2
chardet            3.0.4
click              7.1.2
cmsis-pack-manager 0.2.10
colorama           0.4.3
cryptography       3.1
fasteners          0.15
future             0.18.2
hidapi             0.9.0.post3
icetea             2.0.1
idna               2.10
intelhex           2.2.1
Jinja2             2.11.2
jsonmerge          1.7.0
jsonschema         3.2.0
junit-xml          1.9
lockfile           0.12.2
manifest-tool      2.0.0
MarkupSafe         1.1.1
mbed-cli           1.10.4
mbed-cloud-sdk     2.2.0
mbed-flasher       0.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.12
mbed-os-tools      0.0.15
milksnake          0.1.5
monotonic          1.5
pip                20.2.2
prettytable        0.7.2
psutil             5.7.2
pyasn1             0.3.1
pycparser          2.20
pycryptodome       3.9.8
pydash             4.8.0
pyelftools         0.26
pyparsing          2.4.7
pyrsistent         0.16.0
pyserial           3.4
python-dateutil    2.8.1
python-dotenv      0.14.0
pytz               2020.1
pyusb              1.0.2
pywin32            228
PyYAML             5.3.1
requests           2.24.0
semver             2.10.2
setuptools         47.1.0
six                1.15.0
soupsieve          2.0.1
transitions        0.8.2
urllib3            1.25.10
WMI                1.5.1
yattag             1.14.0

#### How is this defect reproduced ?

AGlass0fMilk, 2020-09-11:
By default Mbed uses an implementation of `printf` with reduced functionality called `minimal-printf`. You may need to switch to using the standard `printf` implementation to enable support for the format you're trying to use.

See this:

https://github.com/ARMmbed/mbed-os/blob/master/platform/source/minimal-printf/README.md

You can try adding this to your mbed_app.json:

```
    "target_overrides": {
        "*": {
            "target.printf_lib": "std"
        }
    }
```

0xc0170, 2020-09-14:
Thanks @AGlass0fMilk . I'll close this as resolved.

closed 2020-09-14 by 0xc0170

fengshengchang, 2020-09-21:
@AGlass0fMilk I've tried: "target.printf_lib": "std"  , and The problem has been solved.

But, the following demo code is still error：

`int main()
{
printf("data:%lld \r\n", 22);
}`

result is:
data:ld

I can't get the correct result.

kjbracey-arm, 2020-09-22:
Looks to me like you've managed to end up using the newlib-nano C library, which I think is the default for a GCC bare-metal build. That only supports C90 formats, so no long long, sadly.

I believe you can switch that by setting `"target.c_lib"` to `"std"` in the same way.

That would be a lot bigger overall though. For size, you might be better off going back to minimal-printf, if you pick up this pending PR which is adding leading-zero support: https://github.com/ARMmbed/mbed-os/pull/13548 

Final note - your example there isn't valid, `22` isn't a `long long`. I'd hope you're getting a build warning for that.