ARMmbed/mbed-os #13603: Error appered whenever I am importing any library with CRC in it.

raj-priyanshu, 2020-09-12:
Hi,  I am trying to use one wire sensors with Nucleo F446RE on the online compiler of mbed os. for this, I have tried these two libraries.
https://os.mbed.com/users/simonbarker/code/Onewire/
https://os.mbed.com/users/richardlane/code/DS18B20_1wire/
but the compiler is raising an error of missing ")" before CRC. but I went through both library and I didn't show any parenthesis mismatch. I am very new to mbed os. 

### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
It can be reproduced by just importing any of these two library.

raj-priyanshu, 2020-09-12:
Hi, ciarmcom,
I apologize for not following the format. But I am very new on this platform and as I am using the online mbed compiler. I don't have much idea about those details I am just started on this platform and still try to understand this new environment. I just import mbed from the importing section soI also have no idea about the version.

0xc0170, 2020-09-14:
What Mbed OS version are you using?

2 libraries you shared are from 2013 (they were tested most likely with Mbed 2 library).  these are not official maintained libraries, please ask on the forum. 

I'll close this one.

closed 2020-09-14 by 0xc0170