ARMmbed/mbed-os #13616: Is it acceptable to create a PR for mbed-os-5.15.x versions

mbedNoobNinja, 2020-09-15:
### Description of defect

I thought that #13125 is the last version but suddenly an #13381 appeared!

If PR are still accepted, how should I structure them if I want to add support for:

- components: ["**`SD`**", "**`FLASHIAP`**"]

- device_has: ["**`USBDEVICE`**", "**`FLASH`**", "**`LPTICKER`**"]

- bootloader_supported [:**`true`**]

SD will be with external lib, (probably with the name SDHI), because it does not use SPI bus, but the native SDHI controller, similar to some of the Nuvoton boards!

#### Target(s) affected by this defect ?

`VK_RZ_A1H`

#### Toolchain(s) (name and version) displaying this defect ?

`ARM`, `GCC_ARM`, `IAR `

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-5.15.x` 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-09-15:
> I thought that #13125 is the last version but suddenly an #13381 appeared!

Yes, currently 5.15 is still opened. 

What do you want to add ? Is it a target update ? The 5.15 branch is only for patch updates.

mbedNoobNinja, 2020-09-15:
Target update

- For now `SD`

- In the near future the other mentioned above.

0xc0170, 2020-09-15:
Yes, ti should not be a problem. I'll close this question.

closed 2020-09-15 by 0xc0170