ARMmbed/mbed-os #13624: STM32WB55, mesh API - Zigbee and Thread protocol

pawelm87, 2020-09-16:
### Description of defect

My issue is kind of question/problem with the mesh api. Please don't delete it, I have not found place to describe it.

I want implement Mesh network (Zigbee or Thread protocols) based on STM32WB55 because ST gives all stacks. To do it I need mesh api for Thread and ZigBee for test. 

Mesh API on the mbed site is not available. The below links redirect to 404 error page:
From https://os.mbed.com/docs/mbed-os/v6.3/apis/6LoWPAN-ND-tech.htmlpage 
To connect to the 6LoWPAN-ND network, use the [6LoWPAN-ND interface API](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h).
From the information on this page it appears that 6LoWPAN-ND interface is the ZIgBEE protocol.

From page https://os.mbed.com/docs/mbed-os/v6.3/apis/wisun-tech.html
To connect to the Wi-SUN network, use the [Wi-SUN interface API](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/mbed-mesh-api/mbed-mesh-api/WisunInterface.h).

Where can I find API for Thread protocol ? MeshInterface hierarchy refers to 3 mesh interfaces, also to ThreadInterface, but nowhere is not documentation, api, anything.
[image]

#### Target(s) affected by this defect ?
STM32WB55

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 1.2.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 5.14 and mbed-os 6.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
The latest

#### How is this defect reproduced ? 
None

pawelm87, 2020-09-17:
@ciarmcom I updated what you asked for.

0xc0170, 2020-09-17:
@artokin would you be able to help?

artokin, 2020-09-18:
> Mesh API on the mbed site is not available. The below links redirect to 404 error page:
> From https://os.mbed.com/docs/mbed-os/v6.3/apis/6LoWPAN-ND-tech.htmlpage

There seems to be additional "page" in the URL, the correct URL would be:
https://os.mbed.com/docs/mbed-os/v6.3/apis/6LoWPAN-ND-tech.html

TuomoHautamaki, 2020-09-18:
I'll clarify the question related to Thread. I'll get back to this..

pawelm87, 2020-09-18:
@artokin my mistake but this is not the issues, on page is link to API to 6LoWPAN-ND protocol and this link redirect to 404 error page

artokin, 2020-09-18:
I'm sorry, the directory structure has changed in Mbed OS 6 and those links are pointing to the older Mbed OS (mbed-os-5.15). Those files are now located in : https://github.com/ARMmbed/mbed-os/tree/master/connectivity/nanostack/mbed-mesh-api/mbed-mesh-api

We will fix the documentation in the near future.

pawelm87, 2020-09-18:
@artokin thank you for help,
Will these API work with STM32WB55 which has official ZigBee and Thread stack ?

I found a few example dedicated for mesh network :
https://github.com/ARMmbed/mbed-os-example-mesh-minimal
https://os.mbed.com/docs/mbed-os/v5.14/apis/mesh-api.html
https://github.com/ARMmbed/nanostack-border-router

But in above examples API was usage in different way :(

artokin, 2020-09-21:
We have not used STM32WB55 in our development. I'm not sure if Nanostack RF drivers (https://github.com/ARMmbed/mbed-os/tree/master/connectivity/drivers/802.15.4_RF) can directly work with that board.

nanostack-border-router is using nanostack (https://github.com/ARMmbed/mbed-os/tree/master/connectivity/nanostack/sal-stack-nanostack/nanostack) C-API's directly. The purpose of mbed-mesh-api is to hide complexity of Nanostack API's so that applications are easier to build. Therefore mbed-os-example-mesh-minimal is using mbed-mesh-api that is written in C++ and is also documented Mbed OS docs.

pawelm87, 2020-09-21:
unfortunately example-mesh-minimal not work with STM32WB55:

Error: L6218E: Undefined symbol NanostackRfPhy::get_default_instance() (referred from BUILD/NUCLEO_WB55RG/ARMC6/mbed-os/features/nanostack/mbed-mesh-api/source/LoWPANNDInterface.o).

@artokin or @LMESTM whether there are plans to add NanostackRfPhy support for this chip by ST or mbed ?

TuomoHautamaki, 2020-09-30:
@pawelm87 my apologies for the slow responses and I still need to pass this further to the right people.
@andypowers Pawel is having questions related to mbed OS and Thread connectivity protocols support. Can you check and comment. Thanks