ARMmbed/mbed-os #13628: STM32WB55 mac address always change

Marchel-Ace, 2020-09-17:
### Description of defect

My issue is kind of question/problem with the GAP API, i try to running GAP example but everytime it change mode, the mac address changing but when i try the example in NRF52 the mac address is remain static, I want to print static/public ble address to identify device

#### Target(s) affected by this defect ?

STM32WB55

#### Toolchain(s) (name and version) displaying this defect ?

Mbed Studio: 1.2.1

#### What version of Mbed-os are you using (tag or sha) ?

Mbed os 6.21 and Mbed Os 6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Latest

#### How is this defect reproduced ? 

None

0xc0170, 2020-09-17:
cc @ARMmbed/mbed-os-pan @ARMmbed/team-st-mcd

pan-, 2020-09-17:
@Marchel-Ace The STM32WB55 doesn't contain a valid public MAC address, it's default value is `FF:FF:FF:FF:FF:FF`. This is the case for other controllers, as they do not all possess a public MAC address available out of the box. For these systems public MAC addresses are usually assigned in factory. 

To solve that issue we use, by default, a random static MAC address. The Bluetooth spec mandates to use a single Random static address until the next power cycle; an implementation may use a different random static address after the controller has been powered off. 
For ST controllers, a new random static address is generated every time the controller is reset. This explains the address changes you are observing. On Nordic, the random static address is constructed from a persistent number unique across all Nordic MCU. 

I'd suggest printing the random static address every time ble is initialized and use that information to identify the device.

Marchel-Ace, 2020-09-17:
thanks for the explanation, but when I set advertise own address type to the public, the advertise mac address remains static even next power cycle, you say "The Bluetooth spec mandates to use a single Random static address until the next power cycle" but it doesn't affect advertise mac address when it set to public, my question is how do I get/print this static advertise mac address?

When I print the mac address using GAP API it shows the random mac address which is correct according to your explanation but the advertised public mac address act a little bit different it still static, I mean is ok when I can't get a factory public mac address which is ST doesn't contain valid public mac address but I can get static mac address from advertising but how do I get this static public  mac address and print to serial, for now, I use ble scanner app to see the advertised public mac address

thank you for your reply

pan-, 2020-09-21:
Ok, I think I understand you issue now. 

What do you think of having one function that returns the device public address in Gap ? We can imagine another function to return the random static one too.

```c++
address_t Gap::getPublicAddress(); 
address_t Gap::getRandomStaticAddress();
``` 

At the same time, it would make sense to deprecate `ble_error_t Gap::getAddress(own_address_type_t &type, address_t& address)` as it implies a single address type is used for all operations which is not always the case. Scanning can be made with random static while advertising is being made with the public address for instance.

jeromecoutant, 2020-10-01:
@pan- @Marchel-Ace 
Any update ?

Marchel-Ace, 2020-10-02:
Thanks for your explanation @pan-, now I know what Iam gonna do with this limitation on stm32, I think my case has been resolved, Thank you for your assistance.

closed 2020-10-02 by Marchel-Ace