ARMmbed/mbed-os #13635: TARGET_STM32F412xG contains system_clock.c in MCU level

boraozgen, 2020-09-18:
### Description of defect

Probably forgotten:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F412xG/system_clock.c 

should be moved to 

https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F412xG/TARGET_NUCLEO_F412ZG

as already is with other targets.

#### Target(s) affected by this defect ?
STM32F412xG

#### Toolchain(s) (name and version) displaying this defect ?
Not relevant.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Not relevant.

#### How is this defect reproduced ? 
Not relevant.

boraozgen, 2020-09-18:
System_clock.c defines board level behavior, such as clock initialization with external or internal clock. For example we use a custom board with this chip where we provide our own system_clock.c implementation which suits our board. Therefore this file should reside in board level.

jeromecoutant, 2020-09-18:
Do you think something like:
````
#if HSE_VALUE==8000000
        RCC_OscInitStruct.PLL.PLLM            = 8;
...
#elif HSE_VALUE==XXX
        RCC_OscInitStruct.PLL.PLLM            = X;
...
#else
error("xxx")
#endif
````

could match your need ?

boraozgen, 2020-09-18:
Well I do not exactly know the exact needs of the configuration since we generated the file from STM32CubeMX. I can check which lines make the difference, but I am sure that the original system_clock.c did not work out of the box.

boraozgen, 2020-09-18:
In this case could the functions be defined as `__weak` to let custom board users to override them?

closed 2020-09-30 by 0xc0170