ARMmbed/mbed-os #13638: mbed deploy command not pulling the right commit SHA

cy-prasanthkumar, 2020-09-18:
### Description of defect

I have "mbed-os.lib" pointing to mbed-os-6.3.0 (https://github.com/ARMmbed/mbed-os/#0db72d0cf26539016efbe38f80d6f2cb7a3d4414). The command "mbed deploy" is not pulling the mbed-os-6.3.0 commit SHA. Instead it always defaults to "latest" tag.

#### Target(s) affected by this defect ?
N/A - Mbed tool issue.

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.3.0
mbed-os.lib: https://github.com/ARMmbed/mbed-os/#0db72d0cf26539016efbe38f80d6f2cb7a3d4414 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli version: 1.10.4

#### How is this defect reproduced ? 
- Create mbed-os.lib pointing to mbed-os-6.3.0 (https://github.com/ARMmbed/mbed-os/#0db72d0cf26539016efbe38f80d6f2cb7a3d4414).
- Deploy mbed-os using "mbed deploy" command.
- Check that the "mbed-os" has checked out with "latest" tag instead of the requested commit SHA.

cy-prasanthkumar, 2020-09-18:
Please note that the "mbed deploy" works fine for mbed-os-6.2.1. So it looks like the issue is only with mbed-os-6.3.0.

cy-prasanthkumar, 2020-09-22:
Quick way to reproduce this issue - **mbed import https://github.com/ARMmbed/mbed-os-example-wifi.** 
Observe that it checks out mbed-os-5.15.4 instead of mbed-os-6.3.0.

adbridge, 2020-10-27:
@cy-prasanthkumar it looks like a 'latest' tag was accidentely pushed back to mbed-os causing that to be pulled rather than the branch by the same name (thus causing your issue). This should be resolved now so will close this. Please let us know if you see it again.

closed 2020-10-27 by adbridge