ARMmbed/mbed-os #13642: Enable SPIF with bare metal profile

masoudr, 2020-09-20:
### Description of defect

As described in [here](https://os.mbed.com/docs/mbed-os/v6.3/bare-metal/index.html) enabling storage is possible with the bare-metal profile. I'm trying to use SPIF with a bare-metal profile but I get linker errors and it seems that it skips the compilation of `SPIFBlockDevice.cpp` file. Maybe I'm missing something in the configuration. Please let me know.

#### Target(s) affected by this defect ?

NucleoF103RB (STM32F103RBT6)

#### Toolchain(s) (name and version) displaying this defect ?

GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-os 6.2.0

#### How is this defect reproduced ? 

`mbed_app.json`:

```
{
    "requires": ["bare-metal"],
    "macros": [
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "target_overrides": {
        "*": {
            "target.c_lib": "small",
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": false,
            "target.clock_source": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
            "target.macros_add": ["HSE_VALUE=8000000"],
            "target.components_add": [
                "SPIF"
            ]
        }
    }
}
```
`main.cpp`:

```
#include "mbed.h"
#include "SPIFBlockDevice.h"

// Create flash device on SPI bus with PTE5 as chip select
SPIFBlockDevice spif(PTE2, PTE4, PTE1, PTE5);

int main()
{
}
```

boraozgen, 2020-09-23:
As explained [here](https://os.mbed.com/docs/mbed-os/v6.3/bare-metal/using-the-bare-metal-profile.html#2-adding-apis), you have to add the respective `mbed_lib.json` names under the `requires` array. Here the name you need is `spif-driver`. Potentially you may need to add further APIs if there are dependencies.

masoudr, 2020-09-23:
@boraozgen Thank you so much. I didn't know that. I ended up the following config:
```
{
    "requires": ["bare-metal", "spif-driver", "mbed-trace"],
    "target_overrides": {
        "*": {
            "target.c_lib": "small",
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": false,
            "target.components_add": [
                "SPIF"
            ]
        }
    }
}
```

closed 2020-09-23 by masoudr