ARMmbed/mbed-os #13647: MAX30003WING INTERFACE WITH MAX32620

satyaveer12345, 2020-09-21:
++ MbedOS Error Info ++
                       Error Status: 0x80FF0144 Code: 324 Module: 255
                                                                     Error Message: Assertion failed: (SPIName)spi != (SPIName)NC
                                                                                                                                 Location: 0x550B
                                                                                                                                                 File: /extras/mbed-os.lib/targets/TARGET_Maxim/TARGET_MAX32620C/spi_api.c+63
                                                     Error Value: 0x0
                                                                     Current Thread: main  Id: 0x20001C04 Entry: 0x589D StackSize: 0x1000 StackMem: 0x20000608 SP: 0x20001324
      For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt="MAX32620"
                                                                                    -- MbedOS Error Info --

0xc0170, 2020-09-21:
hi, I have to close this as it does not contain the issue template neither the full report.

If you look at the error, find out what goes into `targets/TARGET_Maxim/TARGET_MAX32620C/spi_api.c`, line 63. It seems one of the pins is NC and it should not be. review the function `spi_init`.

closed 2020-09-21 by 0xc0170

satyaveer12345, 2020-09-21:
in this syntax (SPIName)spi != (SPIName)NC use this 
but suppose i use this (SPIName)spi == (SPIName)NC
then what is the means?

satyaveer12345, 2020-09-22:
Thank you so much.

-- 

Satyaveer Kushwaha

Embedded Firmware Trainer

USA +1 (415) 851-7851

India +91 79 23299559

ISO 9001:2015 Company

Teksun Inc
California | Denver | New York

Website: www.teksun.com

Product Design Engineering | Hardware| Firmware | Application Software |
Cloud | Web

Mobile | IoT | Big Data | AI | Prototype | Production | Feasibility Study |
Research | Product

Roadmap | Automotive | Wearable | Tracking | Healthcare | Semiconductor |
Embedded

The content of this email is confidential for the recipient specified in
the message only. It is strictly forbidden to share any part of this
message with any part of this message with any third party without consent
of the sender. If you received this message by mistake, please reply to
this message and follow with its deletion, so that we can ensure such a
mistake does not occur in the future.