ARMmbed/mbed-os #13648: Remove nanostack dependency in netsocket

jeromecoutant, 2020-09-21:
### Description of defect

Hi

Compilation is failing if we remove nanostack.

https://github.com/ARMmbed/mbed-os/blob/master/connectivity/netsocket/source/NetworkInterface.cpp#L17

=> netsocket can not be compiled without nanostack.

#### Target(s) affected by this defect ?

All
NA

#### Toolchain(s) (name and version) displaying this defect ?

All
NA

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ?

0xc0170, 2020-09-21:
This is using ns list from nanostack libservice - as I understand this library is a collection of functionality that is stack agnostic. Or is it not?

@ARMmbed/mbed-os-ipcore @kjbracey-arm

jeromecoutant, 2020-09-23:
Proposition:
could we use MBED_CONF_NANOSTACK_LIBSERVICE_PRESENT macro in netsocket ?

kjbracey-arm, 2020-09-23:
Again, this is inventing a problem. The whole point of libservice (in the environment where it was created) was that it was the utterly foundational utility library for all components in the framework (mbed-client, Nanostack, applications).

Trying to exclude the lowest-level utility library from a build is just making work for yourself. Is there really any point in excluding it from a bare metal Mbed OS build?

kjbracey-arm, 2020-09-23:
> This is using ns list from nanostack libservice - as I understand this library is a collection of functionality that is stack agnostic. Or is it not?

Correct. It's general standalone utilities. Historically associated with Nanostack, and required by it, but no dependency on it. (Or on anything else, with the possible exception of a bit of grief with random number seeding, but I think that's already covered by the Mbed OS porting).

For ns_list in particular, it's the best list implementation in the tree (IMHO, having spent a long time on it 4 years ago), so not using it would be daft. I think more people should be, but there has been general reluctance to do so, partly because its different heritage means it doesn't quite fit Mbed OS style, and hasn't been properly integrated into the docs or anything. (It does have good Doxygen though - not sure if that's reached the Mbed OS docs)

I'd suggest including libservice in the base build, or just adding the dependency to netsocket and anyone else using it.

pea-pod, 2020-09-28:
Just my two cents, but while I do overall like the directory structure updates, I did not understand what ```nanostack-libservice``` was until this issue was written up. 

I now understand that is essentially a number of functions (and data structures) that a socket based application or library would use, so it is here in a common area. 

My belief is that naming it ```nanostack-anything-here``` is confusing as it makes it appear tied directly to nanostack instead of being a socket/IP utility library, like I believe it is. So, if I wanted to write a driver layer, for example, for one of the WizNet TCP/IP on chip devices they make, I would have avoided using this and written some of my own. (I've started and stopped this very thing at least twice and just now realized that this was a general use library.)

My suggestion: rename the folder and at least some items to something like this: ```netsocket-libraries```, ```connectivity-libraries```, or ```net-libs```.

0xc0170, 2020-09-29:
I had similar issue recently with CMake , and realized the same. I initally thought netsocket is pulling in nanostack but no, nanostack-libservice  is just generic library without any other dependencies.

Naming suggestion sounds fine to me but as it came from outside (upstreamed), it might be not that simple. I'll let @ARMmbed/mbed-os-ipcore to review this.