ARMmbed/mbed-os #13650: STM32H7: Unable to access full external SRAM

felipeLeast, 2020-09-21:
### Description of defect
Hello everyone, while compiling my code for STM32H747 I've realized that the Mbed is using the 240MHz clock processor and are not accessing and allowing to allocated static memory over 512Mb. But the discovery board for STM32H747I has an external memory of 256Mbit.

I don't know if this is just a configuration on .ld file or if it's something more complex to implement.

#### Target(s) affected by this defect ?
DISCO_H747I

#### Toolchain(s) (name and version) displaying this defect ?
arm-gcc-eabi 9.3.0

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 

@jeromecoutant

jeromecoutant, 2020-09-23:
Hi
Yes SDRAM is not supported by default in mbed-os.
You could have a look on Cube package:
https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/BSP/STM32H747I-DISCO/stm32h747i_discovery_sdram.c
https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H747I-DISCO/Examples/BSP/CM7/Src/sdram.c

AGlass0fMilk, 2020-09-23:
Hi @felipeLeast 

One of my side projects is a digital gauge cluster based on Mbed OS that uses a dual-core STM32H745 along with an external SDRAM:

https://forums.mbed.com/t/marquesas-mbed-based-open-source-digital-gauge-cluster/8410

My BSP for this board shows how I set up the linker script and where the code to initialize the SDRAM can go in the Mbed boot sequence.

You can find the BSP here: https://github.com/AGlass0fMilk/marquesas-bsp

There is a function for all STM32 targets called `TargetBSP_Init`. It was introduced for this reason -- to allow you to initialize any board-level resources early on in the boot sequence (such as external SDRAM, etc). You can find it in Mbed's sources here:

https://github.com/ARMmbed/mbed-os/blob/59df4efaacba6f0c2d2ce2afcbd6602572a24827/targets/TARGET_STM/mbed_overrides.c#L34-L47

My Marquesas BSP initializes the external SDRAM in this function. You can see the initialization code I used here:
https://github.com/AGlass0fMilk/marquesas-bsp/blob/master/sdram/sdram_ext.c

You will probably need to change the settings and pins used, but most of that code was generated by STM32CubeIDE. You should be able to reference the examples mentioned by @jeromecoutant above to get the code required for your board.

The .ld file has some changes to map out the external RAM but it's not really necessary unless you want the linker to manage where your variables get placed in RAM. In my case, I use the external RAM as a display buffer and for caching bitmaps so I just use explicit pointers to the external SDRAM memory space.

I hope this helps!

jeromecoutant, 2020-10-01:
Could this issue be closed ?
Thx

closed 2020-10-05 by adbridge