ARMmbed/mbed-os #13657: Bare metal: Adding TDBStore API creates many unrelated dependencies

boraozgen, 2020-09-23:
### Description of defect

I presume this was introduced with the 6.3.0 restructuring, since it does not occur in 6.2.0.

I use the Storage API, specifically TDBStore in bare metal profile. With 6.2.0 and earlier, I added `tdbstore` to the `requires` array and it worked with a couple other dependencies. With 6.3.0, the TDBStore API was moved under `kv-global-api`. The issue is, since there are other source files such as `FileSystemStore.cpp` under `kv-global-api`, it creates a chain of dependencies which is almost impossible to satisfy. This is the compiler error I receive when I only add `kv-global-api`:

```
Compile [ 41.9%]: FileSystemStore.cpp
[Fatal Error] FileSystemStore.h@22,10: filesystem/FileSystem.h: No such file or directory
[ERROR] In file included from .\mbed-os\storage\kvstore\source\FileSystemStore.cpp:19:
./mbed-os/storage/kvstore/include/kvstore/FileSystemStore.h:22:10: fatal error: filesystem/FileSystem.h: No such file or directory
   22 | #include "filesystem/FileSystem.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Of course I should not be required to add `filesystem` to my dependencies since I do not use it at all. However when I try to do it, it creates further dependencies such as `littlefs`. It goes on like this.

I think the key problem here is `FileSystemStore.cpp` residing under the same API as KVStore.

#### Target(s) affected by this defect ?
All targets.

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9-2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Create a bare-metal example.
Try `#include "TDBStore.h"`.

boraozgen, 2020-09-23:
Filesystem is included also in `KVMap.h`, which is also required by `kvstore_global_api.cpp`.

Furthermore, the include path `#include "kv_config/kv_config.h"` in `kvstore_global_api.cpp` cannot be resolved in bare metal.

I believe the whole new structure should be tested with bare metal profile in general.

LDong-Arm, 2020-11-11:
I'm in favour of (partially) reverting to pre- Mbed OS 6.3 structure - having fine-grained control of which type of KVStore to enable. Working in progress on this.

LDong-Arm, 2020-11-13:
@boraozgen I've created a PR to improve this: #13908 
Please feel free to have a try, by removing `"kv-global-api"` from `"requires"` and adding `"kvstore", "tdbstore"` and dependencies of TDBStore (e.g. `"blockdevice"`).

closed 2020-12-11 by adbridge

Arachnid, 2022-01-06:
I'm experiencing this issue in mbed-os 6.15.1. `kv-config` and `kv-global-api` both depend on `filesystem`, and so I get errors like the following when compiling using baremetal:
```
./mbed-os/storage/kvstore/kvstore_global_api/include/kvstore_global_api/KVMap.h:23:10: fatal error: 'filesystem/FileSystem.h' file not found
#include "filesystem/FileSystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

0xc0170, 2022-01-10:
@Arachnid We need more details to be able to help you. What `requires` did you enable?
Did you follow Migration guide  from https://github.com/ARMmbed/mbed-os/pull/13908 (it is part of the release notes for 6.6.0 release).

Arachnid, 2022-01-10:
@0xc0170 Yes, I followed those instructions. To reproduce:

1. Create a new "mbed-os-example-blinky-baremetal" project
2. Replace main.cpp with the source from the kvstore example
3. Delete mentions of 'devicekey' from the sample
4. Set `requires` in `mbed_app.json` to `["bare-metal", "kvstore", "tdbstore", "kv-global-api", "kv-config"]`
5. Build

Results in this error:
```
Compile [ 52.0%]: kv_config.cpp
[Fatal Error] KVMap.h@23,10: 'filesystem/FileSystem.h' file not found
[ERROR] In file included from ./mbed-os/storage/kvstore/kv_config/source/kv_config.cpp:19:
./mbed-os/storage/kvstore/kvstore_global_api/include/kvstore_global_api/KVMap.h:23:10: fatal error: 'filesystem/FileSystem.h' file not found
#include "filesystem/FileSystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

As you can see from the source, `kv_config.cpp` unconditionally imports `KVMap.h`, which unconditionally imports `filesystem/FileSystem.h` - so any attempt to enable the kv config module will pull in the entire filesystem hierarcy.

Arachnid, 2022-01-24:
Any news on this?

PR #13908 (2020-12-11): Restructure KVStore to one library per store type

### Summary of changes 

Fixes: #13657 
Mutually depends on: https://github.com/ARMmbed/mbed-os-example-kvstore/pull/60

In #13307 we combined multiple `KVStore` types (`TDBStore`, `FileSystemStore`) and the global API into one library, `kv-global-api`. This is inconvenient for users of bare metal: if they only want one store type, they need to add dependencies for other unused types and the global API to `"requires"`. This becomes cumbersome as `FileSystem` has many dependencies.

This PR restructures KVStore to follow the same logic of [Bloc [... cut]