ARMmbed/mbed-os #13675: NUCLEO_G031K8: pinmap not found for peripheral

pilotak, 2020-09-28:
### Description of defect
When you inherit the `NUCLEO_G031K8` in `custom_targets.json` and you redirect the STDIO console to other pins you can't use the LPUART nor the ADC as well as SPI on pins `PA_2` and `PA_3` because those definitions are commented out.

At the moment it throws up an error
```
++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Module: 1
Error Message: pinmap not found for peripheral
```

It should **not** be commented out as all other STM32 targets have those pins enabled even though you will not use such a function due to console. The only pins commented out are the one that uses TIMER for `us_ticker`

@AGlass0fMilk 

https://github.com/ARMmbed/mbed-os/blob/f38aa597c8669c6bf32c240e0507a174804715c0/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TARGET_NUCLEO_G031K8/PeripheralPins.c#L45-L46

https://github.com/ARMmbed/mbed-os/blob/f38aa597c8669c6bf32c240e0507a174804715c0/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TARGET_NUCLEO_G031K8/PeripheralPins.c#L135

https://github.com/ARMmbed/mbed-os/blob/f38aa597c8669c6bf32c240e0507a174804715c0/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TARGET_NUCLEO_G031K8/PeripheralPins.c#L145

https://github.com/ARMmbed/mbed-os/blob/f38aa597c8669c6bf32c240e0507a174804715c0/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TARGET_NUCLEO_G031K8/PeripheralPins.c#L172

https://github.com/ARMmbed/mbed-os/blob/f38aa597c8669c6bf32c240e0507a174804715c0/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TARGET_NUCLEO_G031K8/PeripheralPins.c#L184

#### Target(s) affected by this defect ?
NUCLEO_G031K8

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM
#### What version of Mbed-os are you using (tag or sha) ?

f38aa597c8669c6bf32c240e0507a174804715c0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-clli

#### How is this defect reproduced ? 

`custom_targets.json`
```json
{
  "TEST": {
    "inherits": ["NUCLEO_G031K8"]
  }
}
```

`main.cpp`
```cpp
BufferedSerial test(PA_2_ALT0, PA_3_ALT0);
```

`mbed_app.json`
```json
{
    "target_overrides": {
        "TEST": {
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10",
        }
    }
}
```

jeromecoutant, 2020-09-29:
Hi

> It should not be commented out as all other STM32 targets have those pins enabled even though you will not use such a function due to console.

Yes, we comment default console pins in all tables to avoid application to use these pins as ADC for ex...
This avoid many issues in GitHub :-)
In case of STDIO redirection, we then proposed to define pinmap table as weak.
You can redefine it in local application?

Tell us if this solution doesn't fit your request?

AGlass0fMilk, 2020-09-30:
As @jeromecoutant mentioned, these tables are already tagged with the `MBED_WEAK` preprocessor. This allows you to simply copy paste the file, uncomment the commented out lines as you require, and remove the `MBED_WEAK` declaration from your file's implementation.

You can even place this new .c file in a folder such as `TARGET_MY_CUSTOM_G031K8_TARGET` and it will only override the default tables when you build for that target!

Please close this issue if this answers your question.

pilotak, 2020-10-05:
Ok, makes sense. Thank you

closed 2020-10-05 by pilotak