ARMmbed/mbed-os #13676: DHCP not functioning - KSZ8794 (Ethernet Switch) implementation with Kinetis MK64 (FRDM-64 based custom hardware)

sklatham69, 2020-09-28:
### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

sklatham69, 2020-09-28:
We are using the MK64FN1M0VLL12 and the KSZ8794 4-port Ethernet Switch.  Using Mbed Studio for development.  Mbed v6.3.  More details to come shortly.  Again, we have the same hardware as our V1 except that we changed the PHY chip (KSZ8081) for the Ethernet Switch (KSZ8794) and changed control from MDIO/MDC to SPI.  Mbed OS works fine with V1 but V4 with switch vs PHY doesn't work.

0xc0170, 2020-09-29:
@sklatham69 can you edit the first comment and fill in all details required for a technical bug. Thanks.

Have you made any debugging/exploring yet to find out what is exactly is failing (which object) ?

sklatham69, 2020-09-29:
Thanks for the comment.  It looks like we found it but we are testing today.  This has been a bear for us.

It appears, since Mbed was designed to support the FRDM-64 assembly, that the RXER line is incorporated into the MAC interface.  This isn't required for RMII MAC mode (KSZ8794) but for RMII PHY Mode (KSZ8081 used on FRDM-64).  

So, we have RXER connected in our V1 and not the V4 per the KSZ8794 connection chart.

We built our previous firmware version for V4 from scratch using CoOS and, obviously, didn't incorporate the RXER.

We simply overlooked this interface issue during our Mbed port and put off the solution to do other things.  We assumed it to be some other, more complicated, issue.

We are validating that this is the issue today and I will close this afterwards.

sklatham69, 2020-09-30:
Looks like our connection issue is resolved.  Mbed was looking at the RXER line (that is used for MII and SW-to-PHY connections (not used in SW-to-MAC connections).  We disabled this in Mbed and connection is now up.

0xc0170, 2020-09-30:
Good to hear this is fixed 👍  Shall we close this issue?

sklatham69, 2020-09-30:
Yes ... thanks

closed 2020-09-30 by sklatham69