ARMmbed/mbed-os #13678: No way to change hostname of device for dhcp.

Spinnaker-design, 2020-09-28:
### Description of defect

There is no way to set the hostname of the device when using the Ethernet interface or similar interfaces.

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
Mbed 6.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio

#### How is this defect reproduced ? 

N/A

0xc0170, 2020-09-29:
cc @ARMmbed/mbed-os-ipcore

Spinnaker-design, 2020-10-13:
I believe it has been updated correctly.

adbridge, 2020-10-15:
@Spinnaker-design Some more information in the description is useful. E.g. what you are trying to do and why.

Spinnaker-design, 2020-10-21:
Hi @adbridge I'm trying the set the hostname in the DHCP request on the network interface.  I can set the mac address, but I cannot set the hostname, so the devices are listed as "unknown" in the device table on my router.

dusan2307, 2020-10-29:
DHCP Client should be able to return MAC and Hostname information to the DHCP Server. See sample from Router DHCP Reservation list. It only returns MAC, not Hostname.
[image]

Spinnaker-design, 2020-12-03:
Hi,

Just wanted to check if you've had a chance to look at this?  Is there a work around option?

dusan2307, 2021-08-13:
Any update on this? This is a relatively standard feature for network devices. DHCP Client to set HOSTNAME.

0xc0170, 2021-08-16:
cc @ARMmbed/mbed-os-connectivity

mdkf, 2022-02-03:
Any update on this?