ARMmbed/mbed-os #13682: Multiple Tickers causing CMSIS-RTOS error: stack overflow

ZaneV, 2020-09-30:
### Description of defect
When printing bytes over USB using the USBSerial library and the _putc function the application crashes with the following fault:

`++ MbedOS Error Info ++  
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x2BBF5
Error Value: 0x1
Current Thread: rtx_idle Id: 0x10001EA0 Entry: 0x2BBC9 StackSize: 0x200 StackMem: 0x0 SP: 0x10007F2C`

After some troubleshooting we have narrowed down the issue to the following: We run 10 Tickers that have no relation to the USB functions we use but are for other scheduling in the application, by removing just one of these Tickers the fault disappears. This failure will occur when we simply define the 10th Ticker and have no callback functions attached to them. i.e 

`Ticker myTicker;`

This will cause the application to crash at some point while using USBSerial _putc function.

#### Target(s) affected by this defect ?
LPC1768

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM Version 9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
tag: mbed-os-5.13.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 

mbed_app.json file:
`{
	"requires": ["bare-metal",
		"sd",
		"filesystem",
		"fat_chan",
		"rtos",
		"rtos-api",
		"usb"],
	"target_overrides": {
		"LPC1768": {
			"target.components": ["SD"],
			"target.features_add": ["STORAGE"],
			"target.mbed_app_start": "0x20080",
            "target.printf_lib": "minimal-printf"
		}
	}
}`

This error is reproduced by initializing 10 tickers (again, no need to even attach a callback) and then push bytes to the USB port via USBSerial _putc command.

0xc0170, 2020-09-30:
cc @ARMmbed/mbed-os-core

0xc0170, 2020-09-30:
What do you think is causing stack overflow ? Have you tried https://os.mbed.com/blog/entry/Tracking-memory-usage-with-Mbed-OS/ ?

ZaneV, 2020-10-13:
@0xc0170 Thanks, we were able to narrow this down to a pointer that was over writing the "magic word" in the RTX Idle Thread struct. Please close this issue as this was our mistake. Thanks again for your help.

closed 2020-10-14 by 0xc0170