ARMmbed/mbed-os #13687: NUCLEO_F303RE with ESP8266 failed to connect WIFI

moshe-shahar, 2020-09-30:
### Description of defect

From mbed-os 6, NUCLEO_F303RE with ESP8266 failed to connect WIFI network.
`wifi->scan` return `NSAPI_ERROR_DEVICE_ERROR`
`wifi->connect` return `NSAPI_ERROR_CONNECTION_TIMEOUT`
The same application builds with `mbed-os-5.15.5` working as expected.

#### Target(s) affected by this defect ?
NUCLEO_F303RE

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced? 

Build https://github.com/ARMmbed/mbed-os-example-wifi for `NUCLEO_F303RE`:
1. Edit mbed_app_esp8266.json with your WIFI credentials
1. Edit mbed_app_esp8266.json to configure esp8266 params:
```
        "NUCLEO_F303RE": {
            "esp8266.debug"                             : true,
            "esp8266.rx"                                : "D2",
            "esp8266.tx"                                : "D8",
            "esp8266.rts"                               : "PA_12",
            "esp8266.rst"                               : "D9",
            "esp8266.cts"                               : "PA_11"
        }
```
3. `mbed compile -t GCC_ARM -m NUCLEO_F303RE --app-config mbed_app_esp8266.json`

Flash and run the application to scan for available networks.

moshe-shahar, 2020-09-30:
Application log with ESP8266 debug output:
```
WiFi example
Mbed OS version 6.3.0

Scan:
AT> AT
AT? OK
%n
AT< rlAT< n

          l`sn                                                                             

              l`rlnAT<                                                                      
                       rloà                                  
                           oâll`ânllAT! ready                                                   
AT? OK
%n
AT<
AT(Timeout)
AT> AT
AT? OK
%n
AT< AT
AT= OK

AT? %n
AT> ATE0
AT? OK
%n
AT< ATE0
AT= OK

AT? %n
AT> AT+UART_CUR=115200,8,1,0,3
AT? OK
%n
AT= OK

AT> AT+GMR
AT? AT version:%*d.%*d.%*d.%*d%n
AT= AT version:1.7.0.0
AT? OK
%n
AT< (Aug 16 2018 00:57:04)
AT< SDK version:3.0.0(d49923c)
AT< compile time:Aug 23 2018 16:58:12
AT< Bin version(Wroom 02):v1.7.0
AT= OK

AT> AT+GMR
AT? SDK version:%*d.%*d.%*d%n
AT< AT version:1.7.0.0(Aug 16 2018 00:57:04)
AT= SDK version:3.0.0
AT? OK
%n
AT< (d49923c)
AT< compile time:Aug 23 2018 16:58:12
AT< Bin version(Wroom 02):v1.7.0
AT= OK

AT? %n
AT> AT+CWMODE_DEF=1
AT? OK
%n
AT= OK

AT> AT+CWCOUNTRY_DEF=0,"CN",1,13
AT? OK
%n
AT= OK

AT> AT+CWCOUNTRY_CUR=0,"CN",1,13
AT? OK
%n
AT= OK

AT> AT+CIPRECVMODE=1
AT? OK
%n
AT= OK

AT> AT+CWMODE_CUR=1
AT? OK
%n
AT= OK

AT> AT+CIPMUX=1
AT? OK
%n
AT= OK

AT> AT+CWLAP=,,,0,0,0
AT? OK
%n
AT! +CWLAP:
AT? (%*d,"%*32[^"]",%*hhd,"%*hhx:%*hhx:%*hhx:%*hhx:%*hhx:%*hhx",%*hhu,%*d,%*d,%*d,%*d,%*d,%*d)
%n
AT< (0,"echonet",-69,"58:8b:f3:99:c2:08",2,-42,0,0,0,7,1)
AT< +CWLAP:(3,"DNA-Mokkula-2G-64657Y",-89,"b0:c1:9e:52:b7:38",6,-59,0,4,4,7,1)
AT< +CWLAP:(3,"systest2",-74,"b4:5d:50:7d:a6:c0",7,-46,0,4,4,7,0)
AT< +CWLAP:(3,"systest",-52,"80:2a:a8:8a:e9:13",10,-49,0,4,4,7,0)
AT< OK
AT(Timeout)
AT(Aborted)
scan() faAT? %n
AT? %n
iled with return value: -3012
No WIFI APs found - can't continue further.

Connecting to systest...
AT> AT+CWJAP_CUR="systest","RaaSysT35T"
AT? OK
%n
AT! WIFI
AT? %*12[^"]
%n
AT= CONNECTED

AT? OK
%n
AT! +CWJAP:
AT? %*d%n
AT= 1
AT? FAIL%n
AT<
AT= FAIL
AT(Aborted)
AT? %n
AT<
AT? %n
AT! WIFI
AT? %*12[^"]
%n
AT= DISCONNECT

AT? %n
AT? %n
AT> AT+CWJAP_CUR="systest","RaaSysT35T"
AT? OK
%n
AT! WIFI
AT? %*12[^"]
%n
AT= CONNECTED

AT? OK
%n
AT! +CWJAP:
AT? %*d%n
AT= 1
AT? FAIL%n
AT<
AT= FAIL
AT(Aborted)
AT? %n
AT<
AT? %n
AT! WIFI
AT? %*12[^"]
%n
AT= DISCONNECT

AT? %n
AT? %n
AT> AT+CWJAP_CUR="systest","RaaSysT35T"
AT? OK
%n
AT! WIFI
AT? %*12[^"]
%n
AT= CONNECTED

AT? OK
%n
AT! +CWJAP:
AT? %*d%n
AT= 1
AT? FAIL%n
AT<
AT= FAIL
AT(Aborted)
AT? %
Connection error: -3017
n
AT<
```

moshe-shahar, 2020-09-30:
The commit that breaks and the line that I susspect: https://github.com/ARMmbed/mbed-os/commit/8e789b0162810f74767a0d79315bed3b8344cdcb#diff-c1cec340f3d2f87fc8998df93acb1578R250

jenia81, 2020-09-30:
It will block PDMC release with mbed-os 6.3.0
@yogpan01 FYI

moshe-shahar, 2020-09-30:
> The commit that breaks and the line that I susspect: [8e789b0#diff-c1cec340f3d2f87fc8998df93acb1578R250](https://github.com/ARMmbed/mbed-os/commit/8e789b0162810f74767a0d79315bed3b8344cdcb#diff-c1cec340f3d2f87fc8998df93acb1578R250)

There are two constructors. The first constructor misses initializing `_dhcp` to `true` as done in the second https://github.com/ARMmbed/mbed-os/commit/8e789b0162810f74767a0d79315bed3b8344cdcb#diff-c1cec340f3d2f87fc8998df93acb1578R125. 
I tested it locally and it fixes the `wifi->connect` but not `wifi->scan`.

0xc0170, 2020-09-30:
This is the PR that might be the offender: https://github.com/ARMmbed/mbed-os/pull/12721

As the change there is about dhcp addition and enabling it by default, what happens if you set dhcp to false (there is a method to disable it - `set_dhcp(false)`) ?

0xc0170, 2020-09-30:
cc @ARMmbed/mbed-os-ipcore

moshe-shahar, 2020-09-30:
@0xc0170 , check my previous comment https://github.com/ARMmbed/mbed-os/issues/13687#issuecomment-701451378

moshe-shahar, 2020-10-01:
The PR that broke the scan is https://github.com/ARMmbed/mbed-os/pull/12960

jeromecoutant, 2020-10-01:
> The PR that broke the scan is #12960

Try to remove "c_lib": "small" line:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1443

moshe-shahar, 2020-10-01:
> > The PR that broke the scan is #12960
> 
> Try to remove "c_lib": "small" line:
> https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1443

Thanks, @jeromecoutant.
Yes, removing `"c_lib": "small"` from the target fixes the scan issue.

But, is it a legitimate fix?

jeromecoutant, 2020-10-01:
Maybe "c_lib" should be force to  "std" in
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/wifi/esp8266-driver/mbed_lib.json
?

Is it possible ?
@ARMmbed/mbed-os-ipcore

jeromecoutant, 2020-10-01:
@evedon

evedon, 2020-10-01:
It should be possible to override `target.c_lib`:
```
"target_overrides": {
        "*": {
            "target.c_lib": "small"
        }
     }
```

evedon, 2020-10-01:
@pan- Could you look into this?

moshe-shahar, 2020-10-01:
But, why it was introduced?
Such change influence all modules. Not just ESP8266.

pan-, 2020-10-01:
@evedon We can issue a short term fix for the module. However a long term one would be to have the default printf that supports all the formats available in the libc. Not just a curated list of what is usually useful for developers. 
We disabled minimal printf for the same reason in the ble testing app.

evedon, 2020-10-01:
A coupe of points:

1- If the issue is that scanf is not implemented by minimal-printf or that some printf formats are not suported by minimal-printf, then we need to override "target.printf_lib" (not "c_lib") and set it to "std". 
We could also force a compilation error in the esp8266 library by checking if MBED_MINIMAL_PRINTF is defined.

> However a long term one would be to have the default printf that supports all the formats available in the libc. Not just a curated list of what is usually useful for developers.
I don't think that it is feasible. We want minimal-printf to be as small as possible so it will never support all the formats available in the standard printf library. 

2- If removing "c_lib": "small" from the target fixed the issue then this points at an issue with newlib-nano.

paul-szczepanek-arm, 2020-10-05:
@evedon I applied your comments to the fix @moshe-shahar can you please help us and check if this PR https://github.com/ARMmbed/mbed-os/pull/13707 solves the problem for you?

moshe-shahar, 2020-10-05:
Compilation error:
```
[ERROR] Attempt to override undefined parameter 'target.printf_lib' in 'library:esp8266[*]'
````
https://github.com/ARMmbed/mbed-os/pull/13707/files#diff-72daba8ef643af857ee7d9ec1b681b23R96

moshe-shahar, 2020-10-05:
```
            "target.printf_lib": "std",
```
Setting this in the app config file pass compilation but not fixed the scan issue.

pan-, 2020-10-05:
@evedon minimal printf is compiled with the build system and contains compile time options to disable or enable features like floating point or 64 bits supports. Why wouldn't it be feasible to add more of these flags ? 
Adding more options to allow existing libraries to work while keeping the size smaller than the libc is still a win for the end user.    

@paul-szczepanek-arm Do you know which format broke the driver ?

evedon, 2020-10-05:
> @evedon minimal printf is compiled with the build system and contains compile time options to disable or enable features like floating point or 64 bits supports. Why wouldn't it be feasible to add [...]

Yes, we can add more flags and increase support where we see a gap in functionality.

paul-szczepanek-arm, 2020-10-05:
Minimal printf is actually OK. But it only provides printing. libc nano versions of printf and scanf appear to be broken. Hard to say which format option breaks scanf.
There are 3 ways of fixing it:
change c_lib to std in targets (cannot be overridden by library, only by app_json)
work around by changing the AT parser to use a workaround for broken scanf option (first need to figure out which one is broken)
find and fix bug in libc nano

moshe-shahar, 2021-02-09:
@paul-szczepanek-arm,
Do you know if there is a bug on that issue in the toolchain?

Please check this issue: https://github.com/ARMmbed/mbed-os/issues/14119.
Although it is ARM toolchain, do you think it's similar?