ARMmbed/mbed-os #13705: Cypress cysecuretools isn't installed automatically

MarceloSalazar, 2020-10-02:
### Description of defect

While testing the CY8CKIT-064B0S2-4343W target on a new/fresh environment, found that it requires using a vendor tool `cysecuretools` to setup the project and provision the board.

This python package isn't installed automatically when installing Mbed OS on the PC and using either Mbed CLI or Mbed Studio.

The solution is to add this package to the list of dependencies:
https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt

#### Target(s) affected by this defect ?

CY8CKIT-064B0S2-4343W

#### Toolchain(s) (name and version) displaying this defect ?

Arm and GCC

#### What version of Mbed-os are you using (tag or sha) ?

fbe04097c71d12a28b996084df0304eace81a484

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

$ mbed --version
1.10.4

#### How is this defect reproduced ? 

Use Mbed CLI (`mbed import mbed-os-example-blinky`) or Mbed Studio to import an application and attempt to compile.

@ARMmbed/team-cypress

0xc0170, 2020-11-12:
@ARMmbed/team-cypress how this dependency should be installed?

LDong-Arm, 2020-12-10:
Related to #14022: Even when cysecuretools is installed, it can't locate imgtool which is auto-installed. It would be good to clarify the entire host setup requirements.

LDong-Arm, 2021-02-23:
Unfortunately we can't auto install cysecuretools because it requires python 3.6 or above, whereas Mbed OS's CI checks for Python 2.7 and 3.5 compatibility (due to Mbed CLI 1).

MarceloSalazar, 2021-05-11:
We agreed to close this, as we'd have two steps post build process for installing secure tools

closed 2021-05-11 by MarceloSalazar