ARMmbed/mbed-os #13710: QSPIF is giving error -0x4002 (Parse SFDP Headers Failed)

ranjitsinh-rathod, 2020-10-05:
### Description of defect

#### Target(s) affected by this defect ?
custom hardware based on PSoC 6 Wi-Fi BT Prototyping Kit (CY8CPROTO-062-4343W)

#### Toolchain(s) (name and version) displaying this defect ?
ARM6 using from Mbed Studio

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
ARM Compiler - 6.14
Clangd - 8.0.2
Python tools - 0.5.3
mbed - 1.10.2

#### How is this defect reproduced ? 
1) Initialising QSPI
2) Deinit QSPI
3) Again Init QSPI and got issue in SFDP Header parsing
[QSPIF_IssueInSFDPHeader.txt](https://github.com/ARMmbed/mbed-os/files/5325251/QSPIF_IssueInSFDPHeader.txt)

ranjitsinh-rathod, 2020-10-05:
I am working on custom hardware based on PSoC 6 Wi-Fi BT Prototyping Kit (CY8CPROTO-062-4343W) and working on Mbed 
   OS 5.15.4. I am working on this project for Datahoist.

   In our custom hardware, we have used the S25FL064L chip as a QSPI flash.

   We have an issue in init using QSPIFBlockDevice. When I have enabled traces, I got a print “Init - Verification of SFDP signature 
   and version failed”. Also, a strange thing is this issue comes when we do init -> deinit ->init. So the first time the flash was 
   successfully getting initialized, but after the deinit, when we do init again, it is having issues in SFDP header. I have printed the 
   SFDP header for success as well as failure case, which will be in the log.

0xc0170, 2020-10-15:
cc @ARMmbed/mbed-os-core

adbridge, 2020-10-15:
@ranjitsinh-rathod you are also missing the mbed-os version in the header, could you please add this.

ranjitsinh-rathod, 2020-10-16:
> @ranjitsinh-rathod you are also missing the mbed-os version in the header, could you please add this. Ahh I see you mention it in the description but did not put it in the required field. I'll fix this for

ranjitsinh-rathod, 2020-10-16:
It was closed by mistake. Reopening it. sorry for the inconvenience caused.

0xc0170, 2020-11-12:
cc @ARMmbed/team-cypress

kyle-cypress, 2020-11-12:
@ranjitsinh-rathod , is this issue reproducible on the standard CY8CPROTO-062-4343W? Or only on your custom hardware?

ranjitsinh-rathod, 2020-11-12:
@kyle-cypress no, it is not responsible on the CY8CPROTO-062-4343W kit. But it is always reproducible on the custum board we have designed. Please note that we have used different Qspi chip S25FL064L in our board.