ARMmbed/mbed-os #13711: STM32_gen_PeripheralPins.py takes wrong GPIO file

SDziamchuk, 2020-10-05:
### Description of defect
I try to create STM32H743IITx platform using this guide: https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform
But script STM32_gen_PeripheralPins.py always use GPIO-STM32H74**7**_gpio_v1_0_Modes.xml file instead of GPIO-STM32H74**3**_gpio_v1_0_Modes.xml

Log:
```
python STM32_gen_PeripheralPins.py -m "STM32H743IITx.xml"

Script version 1.12
CubeMX DB version DB.6.0.0

 * Output directory: C:\Users\sdziamchuk\.platformio\packages\framework-mbed\tools\targets\targets_custom\TARGET_STM\TARGET_STM32H743IITx
 * Generating PeripheralPins.c and PinNames.h with 'C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\db\mcu\STM32H743IITx.xml'
 * GPIO file: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\db\mcu\IP\GPIO-STM32H747_gpio_v1_0_Modes.xml
 * I/O pins found: 248 connected: 0
```

#### Target(s) affected by this defect ?
STM32H743IITx

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
STM32_gen_PeripheralPins.py v1.12
CubeMX DB version DB.6.0.0
PlatformIO 5.0.1 (plaform: ST STM32 v8.1.0)

#### How is this defect reproduced ?

jeromecoutant, 2020-10-05:
Well
GPIO_modes file choice came from MCU xml file.
Before opening an internal ticket for CubeMX team, I checked GPIO-STM32H743_gpio_v1_0_Modes.xml, and quickly patched the script in order to use it.

Problem is that results seem not correct...

So I came to the conclusion that:
- file is OK to use
- issue is "only" the file name which should be GPIO-STM32H7_gpio_v1_0_Modes.xml file

jeromecoutant, 2020-10-05:
As #13655 was still in review state, I added a patch :-)

PR #13655 (2020-10-16): STM32_gen_PeripheralPins.py v1.17
### Summary of changes 

Hi

It's a long time I did not sync my personal script with the one in mbed repo.

Many updated lines:
- Octo SPI support
- bug correction
- warning style correction
- new TIM_MST choice algo
- full PinNames.h file creation
- output directory is now
    targets_custom/TARGET_STM/TARGET_STM32XX/TARGET_STM32XXXX

[edit] add correction in GPIO xml files parsing
Fix #13711 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------
closed 2020-10-16 by 0xc0170