ARMmbed/mbed-os #13720: Mbed OS Cypress Bootloader compile Error Help me please

joshheo, 2020-10-06:
### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

joshheo, 2020-10-06:
> ### Description of defect
> #### Target(s) affected by this defect ?
> #### Toolchain(s) (name and version) displaying this defect ?
> #### What version of Mbed-os are you using (tag or sha) ?
> #### What [...]

0xc0170, 2020-10-15:
cc @ARMmbed/team-cypress 

@joshheo have you resolved your issue? The error indicates, there is not enough memory on the device.

This is the condition in the config: `if start > rom_end:` , coming from a function `_generate_bootloader_build` . Verify the sizes are correct for an app and bootloader.

joshheo, 2020-10-15:
Thanks for answering
I'll try it your way

0xc0170, 2020-11-12:
I'll close this as non technical issue (it was answered with some pointers).

closed 2020-11-12 by 0xc0170