ARMmbed/mbed-os #13722: Mbed OS Bootloader compile overlaps section

joshheo, 2020-10-06:
### Description of defect

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-example-bootloader

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
-. targets.jons content
        "device_name": "CY8C6247BZI-D54",
        "mbed_rom_start" : "0x00000000",
        "mbed_rom_size" : "0x20000",
        "mbed_ram_start": "0x08000000",
        "mbed_ram_size": "0x00045800",
        "mbed_app_start " : "0x100000000 " ,
        "mbed_app_size " : " 0x101000000 " ,

-. After compiling, an error occurs as shown below.

c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: section .text LMA [10000000,1000df1f] overlaps section .cy_m0p_image LMA [10000000,10001847]
collect2.exe: error: ld returned 1 exit status
[ERROR] c:/program files (x86)/gnu tools arm embedded/7 2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: section .text LMA [10000000,1000df1f] overlaps section .cy_m0p_image LMA [10000000,10001847]
collect2.exe: error: ld returned 1 exit status

[mbed] Working path "C:\Users\hujun\mbed-os-example-bootloader_201005" (program)
[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\hujun\mbed-os-example-bootloader_201005"
       Command: "C:\Python27\python.exe -u C:\Users\hujun\mbed-os-example-bootloader_201005\mbed-os\tools\make.py -t GCC_ARM -m CY8CKIT_062_WIFI_BT --source . --build .\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM -c"
       Tip: You could retry the last command with "-v" flag for verbose output

I want to know what causes overlaps

0xc0170, 2020-10-15:
cc @ARMmbed/team-cypress

adbridge, 2020-10-15:
@joshheo please update the issue header with the required information. You currently have no description in the description section , no valid version of mbed-os (mbed-os-example-bootloader is an example not a version of mbed-os) and you have not supplied the versions of the tools you are using. Without this information it makes it very hard for anyone to reproduce your issue and thus help you.

joshheo, 2020-10-19:
sorry
Mbed os version is 5.15.

0xc0170, 2020-11-12:
What Gcc version are you using? Although this is rather an error in linker script/defined values in the application.
Please review linker script and if the sizes of regions reported above are correct. There is a region `cy_m0p_image` , right before .text. Review these 2 regions , also the number in the error is same as `"mbed_app_start " : "0x100000000 " ,`

closed 2020-12-16 by joshheo