ARMmbed/mbed-os #13730: Too many warning for forcing conversion to volatile uint32_t reference (Technically this should not be allowed for embedded devices)

arunkumar-mourougappane, 2020-10-07:
### Description of defect
I am seeing too many warnings for conversion to void for a volatile uint32_t reference.

example:
```
Compile [ 84.2%]: FlashIAPBlockDevice.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 84.3%]: lfs_util.c
Compile [ 84.4%]: lfs2_util.c
Compile [ 84.5%]: MBRBlockDevice.cpp
Compile [ 84.6%]: ObservingBlockDevice.cpp
Compile [ 84.7%]: HeapBlockDevice.cpp
Compile [ 84.8%]: SecureStore.cpp
Compile [ 84.9%]: ff.cpp
Compile [ 85.0%]: lfs.c
Compile [ 85.1%]: Dir.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.2%]: LittleFileSystem.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.3%]: File.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.5%]: DirectAccessDevicekey.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.6%]: FileSystem.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.7%]: LittleFileSystem2.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.8%]: FATFileSystem.cpp
[Warning] stm32f4xx_ll_tim.h@2797,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2814,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f4xx_ll_tim.h@2834,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 85.9%]: lfs2.c
Compile [ 86.0%]: stm32f4xx_hal.c
Compile [ 86.1%]: stm32f4xx_hal_can.c
Compile [ 86.2%]: kv_config.cpp
```
#### Target(s) affected by this defect ?
STM32F466RE Target

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9,gcc-arm-none-eabi-6

#### What version of Mbed-os are you using (tag or sha) ?
Tag: mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-CLI

#### How is this defect reproduced? 
The issue was observed even with mbed's official mbed-os-blinky-example with mbed-os-6.3
cloned content using mbed-cli and compiles using `mbed compile -c`

My current config:
```
$ mbed config --list
[mbed] Global config:
ARM_PATH=/home/amouroug/.local/bin/

[mbed] Local config (/home/amouroug/telemetry-rover-controller):
TARGET=NUCLEO_F446RE
TOOLCHAIN=GCC_ARM
TARGET_CODE=0777
ARM_PATH=/home/amouroug/gcc-arm-none-eabi-9/bin

```
My global config uses gcc-arm-none-eabi-6  and I still see the issue. I believe this got introduced with 6.0 release and was not seen in 5.0.

jeromecoutant, 2020-10-07:
Hi
I am sorry but I can't reproduce...

````
$ mbed compile -m NUCLEO_F446RE -t GCC_ARM
...
Compile [ 84.8%]: FATFileSystem.cpp
Compile [ 84.9%]: Dir.cpp
Compile [ 85.0%]: DirectAccessDevicekey.cpp
Compile [ 85.1%]: File.cpp
Compile [ 85.2%]: FileSystem.cpp
Compile [ 85.3%]: LittleFileSystem2.cpp
...
$ arm-none-eabi-g++.exe --version
arm-none-eabi-g++.exe (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
````

arunkumar-mourougappane, 2020-10-08:
I apologize, it seems like the installed version of libstc++ library threw me off. I fixed things on my end and verified build with a 6.0 mbed-os release.

closed 2020-10-08 by arunkumar-mourougappane