ARMmbed/mbed-os #13746: Callback.h@635,9: static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> >::value' "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects"

varichs, 2020-10-09:
### Description of defect

https://os.mbed.com/docs/mbed-os/v6.3/apis/eventqueue.html
use :EventQueue example: deferring from interrupt context
but get error:
[Error] Callback.h@635,9: static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> >::value' "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects"
[ERROR] In file included from .\main.cpp:1:
In file included from ./mbed-os\mbed.h:22:
In file included from ./mbed-os/rtos/include\rtos/rtos.h:29:
In file included from ./mbed-os/rtos/include\rtos/Thread.h:30:
./mbed-os/platform/include/platform\Callback.h:635:9: error: static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> >::value' "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects"
static_assert(std::is_trivially_copyable<F>::value, "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects");
^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./mbed-os/platform/include/platform\Callback.h:382:9: note: in instantiation of function template specialization 'mbed::Callback<void ()>::generate<events::Event<void ()>, void>' requested here
generate(std::move(f));
^
.\main.cpp:38:13: note: in instantiation of function template specialization 'mbed::Callback<void ()>::Callback<events::Event<void ()>, 0>' requested here
sw.fall(queue.event(fall_handler));
^

#### Target(s) affected by this defect ?
NUCLEO-F103RB

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio: 1.2.1

#### How is this defect reproduced ? 
code:
#include "mbed.h"

DigitalOut led1(LED1);
InterruptIn sw(USER_BUTTON);
EventQueue queue(32 * EVENTS_EVENT_SIZE);
Thread t;

// It's safe to use UnbufferedSerial in ISR context
//UnbufferedSerial console(USBTX, USBRX);

void rise_handler(void)
{
//    char buf[64];
    printf("rise_handler in context %p\n", ThisThread::get_id());
//    console.write(buf, strlen(buf));
    // Toggle LED
    led1 = !led1;
}

void fall_handler(void)
{
    printf("rise_handler in context %p\n", ThisThread::get_id());
    // Toggle LED
    led1 = !led1;
}

// main() runs in its own thread in the OS
int main()
{

    // Start the event queue
    t.start(callback(&queue, &EventQueue::dispatch_forever));
    printf("Starting in context %p\r\n", ThisThread::get_id());
    // The 'rise' handler will execute in IRQ context
    sw.rise(rise_handler);
    // The 'fall' handler will execute in the context of thread 't'
    sw.fall(queue.event(fall_handler));

    // while (true) {

    // }
}

AGlass0fMilk, 2020-10-10:
Hi @varichs,

The `ThisThread::get_id()` function call is not expected to work when running inside of an interrupt handler. The application is not running thread mode and therefore there is no thread ID associated with the interrupt context.

varichs, 2020-10-10:
> Hi @varichs,
> 
> The `ThisThread::get_id()` function call is not expected to work when running inside of an interrupt handler. The application is not running thread mode and therefore there is no thread ID associated with the interrupt context.

But this code is an official example. Is this example wrong？https://os.mbed.com/docs/mbed-os/v6.3/apis/eventqueue.html

toyowata, 2020-10-10:
Hi @varichs 

The warning says...
```
./mbed-os/platform/include/platform\Callback.h:635:9: error: static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> >::value' "F must be TriviallyCopyable. Turn on Mbed configuration option 'platform.callback-nontrivial' to use more complex function objects"
```

So, you can try to enable the `platform.callback-nontrivial` in your mbed_app.json.

```
    "target_overrides": {
        "*": {
            "platform.callback-nontrivial"      : true,
```

0xc0170, 2020-10-15:
> So, you can try to enable the platform.callback-nontrivial in your mbed_app.json.

I'll close this as resolved

closed 2020-10-15 by 0xc0170

cbusse1, 2023-04-20:
The documentation should be updated with this note about enabling nontrivial callback functions. 

mbed_app.json:
~~~
{
    "target_overrides": {
        "*": {
            "platform.callback-nontrivial"      : true
        }
    }
}
~~~

harshdobariya, 2024-03-19:
Which mbed_app.json should be updated? There are so many file with the same name.

cbusse1, 2024-03-19:
I mean in the "EventQueue example: deferring from interrupt context" of the EventQueue documentation https://os.mbed.com/docs/mbed-os/v6.16/apis/eventqueue.html
it should be mentioned that the example will only complie/run successfully, if an mbed_app.json is provided with:
~~~
{
    "target_overrides": {
        "*": {
            "platform.callback-nontrivial"      : true
        }
    }
}
~~~
otherwise the error "static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> " will be raised as described above in this issue.

harshdobariya, 2024-03-19:
Sorry, but which mbed_app.json should be updated to platform.callbak-nontrivial : true? 
Cause I tried adding these lines to mbed_app.json of test/config/simple_features and  test/config/app_override_libs but the same error persisted.

harshdobariya, 2024-03-19:
> I mean in the "EventQueue example: deferring from interrupt context" of the EventQueue documentation https://os.mbed.com/docs/mbed-os/v6.16/apis/eventqueue.html it should be mentioned that the example will only compile/run successfully if an mbed_app.json is provided with:
> 
> ```
> {
>     "target_overrides": {
>         "*": {
>             "platform.callback-nontrivial"      : true
>         }
>     }
> }
> ```
> 
> otherwise, the error "static_assert failed due to requirement 'std::is_trivially_copyable<events::Event<void ()> " will be raised as described above in this issue.

cbusse1, 2024-03-19:
You need to create a **new** mbed_app.json file which should be located in the directory of your program (same as main.cpp).

harshdobariya, 2024-03-20:
> You need to create a **new** mbed_app.json file which should be located in the directory of your program (same as main.cpp).

Thank you very much. It worked.