ARMmbed/mbed-os #13748: Enable QSPIF with bare-metal profile

masoudr, 2020-10-09:
### Description of defect

I'm trying to enable QSPIF based on this [guide](https://os.mbed.com/docs/mbed-os/v6.3/apis/qspifblockdevice.html) with a bare-metal profile for my target, but I get errors when I try to compile the code. I'm not sure if I'm missing something in the configuration.

```
[Error] QSPI.h@224,12: field '_qspi' has incomplete type 'qspi_t' {aka 'qspi_s'}
```

#### Target(s) affected by this defect ?
NucleoF103RB (STM32F103RBT6)

#### Toolchain(s) (name and version) displaying this defect ?
GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
`mbed-app.json`:
```
{
    "requires": ["bare-metal", "qspif", "mbed-trace", "events", "drivers-usb"],
    "macros": [
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "target_overrides": {
        "*": {
            "target.c_lib": "small",
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": false,
            "target.clock_source": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
            "target.macros_add": ["HSE_VALUE=8000000"],
            "target.components_add": [
                "QSPIF"
            ],
            "target.device_has_add": ["USBDEVICE", "QSPI"]
        }
    }
}
```
`main.cpp`:
```
#include "mbed.h"
#include "QSPIFBlockDevice.h"

QSPIFBlockDevice qspif(PB_15, PB_14, PC_5, PB_11, PB_13, PB_12);

int main()
{
    printf("qspif test\n");

    // Initialize the SPI flash device, and print the memory layout
    printf("qspif init: %d\n", qspif.init());
    printf("qspif size: %llu\n", qspif.size());
    printf("qspif read size: %llu\n", qspif.get_read_size());
    printf("qspif program size: %llu\n", qspif.get_program_size());
    printf("qspif erase size: %llu\n", qspif.get_erase_size());

    qspif.deinit();
}

```

jeromecoutant, 2020-10-09:
Hi
For me, STM32F103 doesn't support QSPI....?

masoudr, 2020-10-09:
Oops, you are right, I thought it is a software feature. I just read [this](https://os.mbed.com/docs/mbed-os/v6.3/apis/spi-apis.html) and couldn't find out the device support. Anyway, Thanks for the info.

closed 2020-10-09 by masoudr