ARMmbed/mbed-os #13750: Mbed OS BLE stack not allowing application to configure ADV duration timeout

cy-prasanthkumar, 2020-10-12:
### Description of defect

Mbed OS application starts BLE advertisement with timeout value passed as argument as shown below and it expect the BLE advertisement to stop when timeout expires and get an application callback "onAdvertisingEnd()".

```
_ble.gap().startAdvertising(ble::LEGACY_ADVERTISING_HANDLE, test_duration); /* for example, ble::adv_duration_t test_duration(5000) for 5s timeout. */
```

But we observed that BLE advertisement never stopped and did not receive application callback on onAdvertisingEnd(). And, when we debug more into Mbed BLE cordio stack, we found that the ADV duration is set to 0 (which means infinite advertisement which will never time out) as shown below:

```
diff --git a/connectivity/FEATURE_BLE/source/cordio/source/PalGapImpl.cpp b/connectivity/FEATURE_BLE/source/cordio/source/PalGapImpl.cpp
index 07af679582..78e56b8a37 100644
--- a/connectivity/FEATURE_BLE/source/cordio/source/PalGapImpl.cpp
+++ b/connectivity/FEATURE_BLE/source/cordio/source/PalGapImpl.cpp
@@ -208,16 +208,19 @@ ble_error_t PalGap::advertising_enable(bool enable)
             }
         } else {
             uint8_t adv_handles[] = {DM_ADV_HANDLE_DEFAULT};
             uint16_t adv_durations[] = { /* infinite */ 0};
             uint8_t max_ea_events[] = {0};
             DmAdvStart(1, adv_handles, adv_durations, max_ea_events);
         }
```
Could you please clarify us if there is anyway we can override the BLE advertisement duration timeout value from application and get an application callback when timeout expires or is this a bug that needs to be addressed ?

#### Target(s) affected by this defect ?
CYW9P62S1_43438EVB_01

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM v6.3.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli v1.8.3

#### How is this defect reproduced ? 

Start your application with BLE advertisement timeout value set and passed in argument as shown below. Observe that the BLE advertisement never stops.

> ble::adv_duration_t test_duration(5000);
> _ble.gap().startAdvertising(ble::LEGACY_ADVERTISING_HANDLE, test_duration);

pan-, 2020-10-13:
@cy-prasanthkumar Advertising timeout was not supported natively by controllers before Bluetooth 5. We have two paths to enable advertising: 
- Legacy path for 4.X devices. 
- Extended advertising path for 5.x device.  

For legacy advertising, it advertising stops after the timeout passed to `startAdvertising` (an internal timer keep track of it). However reporting of the event to the user was not unified between BT generations (report on BT5 devices, no report on BT4 devices). 

We worked on that area during our effort to bring host based privacy. Behavior should be consistent accross BT versions (see [here](https://github.com/ARMmbed/mbed-os/blob/f32141a05137301eadb8b8143063bd6220d9d679/connectivity/FEATURE_BLE/source/generic/GapImpl.cpp#L2835-L2837)). The feature branch hasn't been merged yet but you can try it here: https://github.com/ARMmbed/mbed-os/pull/13717

cy-prasanthkumar, 2020-12-14:
It does not seem to be resolved yet. I have tested with Mbed OS v6.5.0 (https://github.com/ARMmbed/mbed-os/#f2278567d09b9ae9f4843e1d9d393526b9462783) and using "https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_Advertising". Added the timeout as shown below:
        /* start advertising */
        ble::adv_duration_t test_duration(50000);
        error = _ble.gap().startAdvertising(ble::LEGACY_ADVERTISING_HANDLE, test_duration);

Also, it did not callback the below function which is expected to be called after timeout:
    virtual void onAdvertisingEnd(const ble::AdvertisingEndEvent &event)
    {
        printf("Adv Ended.\n");
    }

pan-, 2020-12-15:
@cy-prasanthkumar I tried to reproduce your issue but it works well with the example. I have few observations: The duration range from `0x1` to `0xFFFF` in steps of 10ms. With the value provided you have to wait more than 10 minutes to experiment the timeout I have reduced it to 10s with the value `1000`. 

Adding the callback to the class is a good first step but you also need to register the event handler into the gap class. In the function `on_init_complete` I added the following line: `_ble.gap().setEventHandler(this);`

cy-prasanthkumar, 2020-12-15:
@pan- You are correct. I verified again with 1000 as timeout value. I can see the ADV stops after 10s and the application callback gets called. Thank you. I am closing this case.

closed 2020-12-15 by cy-prasanthkumar