ARMmbed/mbed-os #13751: DISCO-BL475E-IOT01A doesn't exit deep sleep

alcheagle, 2020-10-12:
### Description of defect
DISCO-BL475E-IOT01A device is unable to exit from deep sleep. 

I wrote a simple code to check power modes of devices and their power consumption. Everything works fine except when the device enters deep sleep.

Here is the sample code:
```cpp
DigitalOut led(LED1);

int main() {
    //sleep_manager_lock_deep_sleep();
    printf("turning on\n");
    printf("System clock is %ld\n", SystemCoreClock);
    
    while (1) {
        printf("\n\ninit\n");
        // blink the LED
        sleep_manager_lock_deep_sleep();
        led=1;
        ThisThread::sleep_for(2s);
        led=0;
        sleep_manager_unlock_deep_sleep();
        
        printf("\n\nrun\n");
        time_t start=time(NULL);
        uint64_t i=0;
        while((time(NULL)-start)<2) {
            i++;
        }
        printf("counter: %llu\n", i);
        printf("\n\nsleep\n");

        sleep_manager_lock_deep_sleep();
        ThisThread::sleep_for(2s);
        sleep_manager_unlock_deep_sleep();
        
        printf("\n\ndeep sleep\n");
        ThisThread::sleep_for(2s);
    }
    return 0;
}
```

the mbed_app.json I am using is:

```json
{
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
        },
        "DISCO_L475VG_IOT01A": {
            "target.clock_source": "USE_PLL_HSI"
        }
    }
}
```

#### Target(s) affected by this defect ?
DISCO-BL475E-IOT01A

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?
ad40b1b2672d59b058f7c104fdc612d0dddf3ac3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

jeromecoutant, 2020-10-12:
Hi
I think we need to apply same patch as in #13725?

alcheagle, 2020-10-12:
Hi,
It could be the same issue. I just noticed I have to change `target.clock_source` to `target.clock-source` in the `mbed_app.json` file.
When setting it with "USE_PLL_HSI" it is working correctly, thus I think the issue depends on the clock configuration.

alcheagle, 2020-10-12:
wait, that is incorrect, the correct setting is ``target.clock_source``. This implies that the ``USE_PLL_HSI`` clock mode is not working.

alcheagle, 2020-10-12:
I tried using the changes applied in #13725. With either ``USE_PLL_HSI`` and ``USE_PLL_MSI`` something strange is happening: after some cycles of the code above the device freezes.

alcheagle, 2020-10-12:
@jeromecoutant can you try the code I posted above on the device (NUCLEO_L476RG) referenced in #13725? I will probably have one by the next week.

I hope the board I am working with is not broken in some way.

alcheagle, 2020-10-12:
The issue may be with ``"target.lpticker_lptim": 1`` setting

jeromecoutant, 2020-10-13:
> I think we need to apply same patch as in #13725?

I confirm same kind of patch is solving the issue.
But why do yo want to change "target.clock_source" ?

alcheagle, 2020-10-14:
I wanted to test whether the issue was present on a single clock configuration. 
However by applying the configuration below, the device freezes after a while.

```json
{
    "config": {
    },
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200
        },
        "DISCO_L475VG_IOT01A": {
            "target.lpticker_lptim": 0
        }
    }
}
```

You can see the device freezes from the image attached below. The graph shows power consumption of the device running the code posted above. After a while the device doesn't resume light sleep state. 

[image]

jeromecoutant, 2020-10-14:
This is not an issue for me. Default configuration with lpticker_lptim chnage only is not supported.

alcheagle, 2020-10-14:
What do you mean? ``lpticker_lptim`` should only be set to 1 because it is not expected to work with stm32l4xx? Or there are other settings to change in order to make it work? I am probably missing some steps.

jeromecoutant, 2020-10-15:
Yes, only lpticker_lptim set to 1 is validated with STM32L4.

alcheagle, 2020-10-15:
Are there any plans to support this feature?

closed 2020-10-20 by 0xc0170

alcheagle, 2020-10-20:
Is it documented somewhere that the device doesn't work when ``lpticker_lptim`` is set to 1? Otherwise I am opening a new issue tracking this.

LMESTM, 2020-10-20:
> Is it documented somewhere that the device doesn't work when `lpticker_lptim` is set to 1? Otherwise I am opening a new issue tracking this.

The targets.json describes each target validated HW configuration, the one that is supported in the master branch.
It allows some re-use between MCU families. It may be modified for optimization and removal of unused drivers for instance, but there is no commitment that all alternative configurations shall work, this is clearly not the case and would not be possible.

What is the feature you want to achieve and why would you like to de-active this option ?

alcheagle, 2020-10-21:
In MCU_STM32L4 it is set to 1, but it doesn't specify that the value 0 is not tested. 

https://github.com/ARMmbed/mbed-os/blob/8ef0a435b2356f8159dea8e427b2935d177309f8/targets/targets.json#L3121-L3124

I am working on projects constrained by power consumption and waking up the device periodically is not a good choice. I know that there is the setting ``lpticker_lptim_clock`` that could allow to less frequent updates, but still the power consumption is higher than it should be.

In stm32l0 family I was able to set this to 0 and the results were promising, but in that case I needed more flash and more RAM, thus the choice to try the l4 family.

LMESTM, 2020-10-21:
> In MCU_STM32L4 it is set to 1, but it doesn't specify that the value 0 is not tested.
> 
> https://github.com/ARMmbed/mbed-os/blob/8ef0a435b2356f8159dea8e427b2935d177309f8/targets/targets.json#L3121-L3124

This could be better documented indeed, I agree. The mbed-os ticker actually requires many very fine grain reprogramming of the next tick. This leads to many issues when trying to implement this with an RTC which is not meant for such usage. That's why you're seeing the product freezing after some time as we most probably run into race conditions when trying to reprogram few µs only after a previous programing was done. I don't think we can easily make it work, not even sure this is actually feasible. It was also very hard to achieve with LPTIM. 

> I am working on projects constrained by power consumption and waking up the device periodically is not a good choice. I know that there is the setting `lpticker_lptim_clock` that could allow to less frequent updates, but still the power consumption is higher than it should be.

I agree that there is a need for allowing long periods of sleep without wake-ups. This would require to suspend the ticker for some time and program RTC alarm for waking-up. I agree this is an area of improvement as it is not yet available. 

> In stm32l0 family I was able to set this to 0 and the results were promising, but in that case I needed more flash and more RAM, thus the choice to try the l4 family.

Unfortunately I think you would face the same instabilities with L0 sooner or later if using RTC as the mbed-os lp ticker.

alcheagle, 2020-10-21:
Thanks for your explanation, I really appreciate it. I understand that this feature is hard to implement, I will consider contributing on this, hoping I am capable of helping.

This is the power consumption of the `NUCLEO_L476RG` while running the code mentioned above, with setting `lpticker_lptim_clock: 4`. The measurements were performed by means of [otii](https://www.qoitech.com/otii/).
[image]

The average was measured with periods of 60s.

Run mode: average 12.5mA
Sleep Mode: 5.99mA 
Deep Sleep Mode: average 6.03uA 
Deep sleep Mode without considering the spikes generated by wakeups: average 2.55uA 
Deep sleep wakeups: 8.24mA that last ~2ms

LMESTM, 2020-10-21:
> Thanks for your explanation, I really appreciate it. I understand that this feature is hard to implement, I will consider contributing on this, hoping I am capable of helping.

That would definitely be of great help ! 
We  can definitely support this initative by reviewing and testing : the lp_ticker and sleep test suites are very good for checking this feature. They should actually report errors when testing with RTC instead of LPTIM.

> The average was measured with periods of 60s.
> 
> Run mode: average 12.5mA
> Sleep Mode: average 5.99mA
> Deep Sleep Mode: average 6.03uA
> Deep sleep Mode without considering the spikes generated by wakeups: average 2.55uA
> Deep sleep wakeups: 8.24mA that last ~2ms

Very interesting. I agree that the impact of the wake-ups here is really high.  One of the reason is the occurence of wake-ups, but also the duration of each wake-ups. One of the reason for this duration is the time it may take to restart the PLL and wait for clock to be stabilized. We have generic implementations here for deep sleep: 
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/sleep.c#L278 
and here for the clock settting (depends on your application):
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/system_clock.c#L107

So depending on your target and final application you may want to fine tune the clock settings and stabilization time, maybe avoid using PLL for instance if this is acceptable, and save time during wake-ups by optimizing this function as well as the clock for your own product. This could noticeably improve the deep sleep average power consumption.

PR #13741 (2020-10-20): STM32L4 update drivers version to CUBE V1.16.0
### Summary of changes 

ST Cube drivers version is updated from V1.14.0 to V1.16.0
which is the latest official version from:

https://github.com/STMicroelectronics/STM32CubeL4/

- directories have been restructured to be aligned with other STM32 families
- Sub-family MCU has been introduced in targets.json to help custom boards use

@ARMmbed/team-st-mcd

#### Impact of changes 

fix #13751

#### Migration actions required 

### Documentation 

-------------------------------------------------------------------------------------