ARMmbed/mbed-os #13761: CoAP blockwise code is out of spec and causes hard fault

Alex-EEE, 2020-10-13:
### Description of defect

In the line of code below, you are calculating the offset using the max block size of the MBED DEVICE, which is WRONG and out of spec (https://tools.ietf.org/html/rfc7959#page-29)

You need to calculate the offset using THE BLOCK SIZE SENT IN THE PACKET.

https://github.com/ARMmbed/mbed-os/blob/05489818e9f6eeebb87233232d77374983e18142/connectivity/libraries/mbed-coap/source/sn_coap_protocol.c#L1283

``
        memcpy(stored_blockwise_payload_ptr->payload_ptr + (block_number * handle->sn_coap_block_data_size), payload_ptr, payload_len);
``

block_number * handle->sn_coap_block_data_size is wrong.

You need to pass the block_size into this function as contained in the incoming back.  (You derive this value already on line 1925 of the same file)  ( uint16_t block_size = 1u << (block_temp + 4);)

#### Target(s) affected by this defect ?
All targets

#### Toolchain(s) (name and version) displaying this defect ?
All toolchains, but I'm using IAR

#### What version of Mbed-os are you using (tag or sha) ?
master (0548981)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Send a blockwise PUT using a smaller blockwise size than is set in SN_COAP_MAX_BLOCKWISE_PAYLOAD_SIZE

0xc0170, 2020-10-14:
@teetak01 would you be able to review if this fix should be upstreamed?

teetak01, 2020-10-14:
Thanks @Alex-EEE 

Looks like a valid issue.

Internal ticket https://armiot.atlassian.net/browse/IOTCLT-4475

ristohuhtala, 2020-10-20:
@Alex-EEE PR to fix this into mbed-os is https://github.com/ARMmbed/mbed-os/pull/13787

0xc0170, 2020-11-12:
Updated via referenced PRs, I'll close as resolved.

closed 2020-11-12 by 0xc0170