ARMmbed/mbed-os #13764: `mbed deploy` deploy incorrect mbed-os revision

moshe-shahar, 2020-10-14:
### Description of defect

Trying to deploy mbed-os-6.3.0 result in mbed-os 5.15.1.
`mbed-cli` mix between `latest` tags and branch:
```
git ls-remote git@github.com:ARMmbed/mbed-os latest*
0db72d0cf26539016efbe38f80d6f2cb7a3d4414        refs/heads/latest
8d797dfa896b42208c91769629da6b9bf5e72182        refs/tags/latest
e642a7d8b3609a7c903e042cd772f00a5d299088        refs/tags/latest^{}
```

#### Target(s) affected by this defect ?
NA

#### Toolchain(s) (name and version) displaying this defect ?
NA

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.4

#### How is this defect reproduced ? 
```
mkdir test-deploy
cd test-deploy/
echo "https://github.com/ARMmbed/mbed-os/#0db72d0cf26539016efbe38f80d6f2cb7a3d4414" > mbed-os.lib
mbed config root .
mbed deploy -vv
```
`mbed-ls` results:
```
---
test-deploy ( revision in the current branch)
`- mbed-os (#e642a7d8b360, tags: latest, mbed-os-5.15.1)
```

adbridge, 2020-10-14:
latest tag has been deleted again. There is currently no satisfactory solution to stop someone from pushing the tag again...

0xc0170, 2020-10-19:
> latest tag has been deleted again. There is currently no satisfactory solution to stop someone from pushing the tag again...

Very unfortunate, I'll check who has write access and will write to all of people involved to remove that tag locally and make sure not pushing it upstream again.

0xc0170, 2020-10-20:
One addition to my msg yesterday: the latest tag was reported being back yesterday. We removed it.

moshe-shahar, 2020-10-21:
@0xc0170 , please reopen that issue until you have a permanent fix for it.

The tag was pushed, again!

0xc0170, 2020-10-21:
@adbridge our scripts needs a fix (the release bot pushed a tag yesterday)

adbridge, 2020-10-21:
> 
> 
> @adbridge our scripts needs a fix (the release bot pushed a tag yesterday)

Not sure how that is even possible as mbed-os would have been cloned from new - but I will investigate just in case

adbridge, 2020-10-21:
So I just tried the steps that @moshe-shahar listed and this is the log I get:

moshe-shahar, 2020-10-21:
No. The tag will return from the cache only if mbed cache is enabled.

adbridge, 2020-10-21:
Looks like there was a config issue is somebody's git setup - now fixed. Hopefully this should not happen again.

0xc0170, 2020-10-26:
I'll close this as resolved. For anyone else looking at this later, don't use git config setting `followTags` set to true for `[push]`. Pushing a branch can cause pushing much more than just  that branch.

closed 2020-10-26 by 0xc0170