ARMmbed/mbed-os #13765: CMake: issues in macro and option files, and the passing of them

LDong-Arm, 2020-10-15:
### Description of defect

In [mbed_set_mbed_target_linker_script()](https://github.com/ARMmbed/mbed-os/blob/feature-cmake/CMakeLists.txt#L112-L123) if you print the values of `${_linker_preprocess_options}` and `${_linker_preprocess_definitions}` with `message()`, and configure an app (e.g. blinky), you'll find those variables are empty. So `compile_time_defs.txt` and `linker_options.txt` are not actually used during preprocessing. There're possibly issues with variable scopes and/or their availability at evaluation time (i.e. function calls).

Even if the above is fixed, there're problems with contents of `compile_time_defs.txt` and `linker_options.txt`:
* macros with spaces in `compile_time_defs.txt` are not properly quoted
* items in `linker_options.txt` are separated by semicolons instead of spaces (GCC only recognises spaces)
* some linking options are not applicable/compatible with the preprocessor

#### Target(s) affected by this defect ?

N/A

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

`feature-cmake` branch

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools 3.4.0
cmake 3.18.4

#### How is this defect reproduced ? 

Apply the following to the `feature-cmake` branch

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 63e3e59a98..2df8eeaa36 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -113,6 +113,7 @@ function(mbed_set_mbed_target_linker_script target)
     get_property(mbed_target_linker_script GLOBAL PROPERTY MBED_TARGET_LINKER_FILE)
     if(MBED_TOOLCHAIN STREQUAL "GCC_ARM")
         mbed_generate_gcc_options_for_linker(${target} _linker_preprocess_definitions _linker_preprocess_options)
+        message(definitions: ${_linker_preprocess_definitions}, options: ${_linker_preprocess_options})
         set(CMAKE_PRE_BUILD_COMMAND
             COMMAND "arm-none-eabi-cpp" -E -P
                 ${_linker_preprocess_options} ${_linker_preprocess_definitions}
Collapse
```

0xc0170, 2020-10-15:
I've fixed part of this on my branch: https://github.com/ARMmbed/mbed-os/commit/5932c7b0c9db3f496b74105c35fbb6dc9ecc46b5

Removed linker options, they should not be there. We should just need definitions. But we might not want all or need to fix some better, see below.

The remaining issue is with config values that are either arrays or strings:
```
-DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_PSK_KEY={0xa0, 0xa1, 0xa2, 0xa3, 0xa4, 0xa5, 0xa6, 0xa7, 0xa8, 0xa9, 0xaa, 0xab, 0xac, 0xad, 0xae, 0xaf} 

-DMBED_CONF_MBED_MESH_API_WISUN_NETWORK_NAME="Wi-SUN Network" 
```

These 2 cause a trouble: `arm-none-eabi-cpp: fatal error: too many input files`. CMake is doing magic with these, I checked Ninja what it contains:

```
-DMBED_CONF_MBED_MESH_API_6LOWPAN_ND_PSK_KEY="{0xa0, 0xa1, 0xa2, 0xa3, 0xa4, 0xa5, 0xa6, 0xa7, 0xa8, 0xa9, 0xaa, 0xab, 0xac, 0xad, 0xae, 0xaf}"

-DMBED_CONF_MBED_MESH_API_WISUN_NETWORK_NAME="\"Wi-SUN Network\"" 
```

@LDong-Arm what values are required for bootloader to pass to linker script? is app size and others part of the config (`MBED_CONFIG_DEFINITIONS` in mbed_config.cmake) ?

LDong-Arm, 2020-10-15:
> @LDong-Arm what values are required for bootloader to pass to linker script? is app size and others part of the config (`MBED_CONFIG_DEFINITIONS` in mbed_config.cmake) ?
> 
> Answer: `MBED_CONF_TARGET_BOOT_STACK_SIZE` is part of config value.

The safest thing to do would be to check macros in GCC_ARM `.ld` files for all targets. There shouldn't be many I hope.

0xc0170, 2020-10-15:
I can finally get the definitions to the linker script, it should be sufficient for the most cases : https://github.com/ARMmbed/mbed-os/pull/13767, please review.

> The safest thing to do would be to check macros in GCC_ARM .ld files for all targets. There shouldn't be many I hope.

Probably just a linker script being used by an app - we know it so we can scan but not certain how to implement it - find all preprocessor lines in the ld file, find out what they use and add these macros to cpp for ld ? I find this very complicated for such simple thing like this, there must be easier way to allow this.

adbridge, 2020-10-15:
@LDong-Arm  for reference 'feature-cmake branch' is not a version of mbed-os and this will be rejected by the issue mirroring script. @0xc0170 is this now addressed and can be closed (ie no need to mirror?)

LDong-Arm, 2020-10-15:
@adbridge Since @0xc0170 already has a PR which I'll review, I'm personally okay with closing this issue and continuing any discussions there.

LDong-Arm, 2020-10-15:
> Probably just a linker script being used by an app - we know it so we can scan but not certain how to implement it - find all preprocessor lines in the ld file, find out what they use and add these [...]

What I mean is, have a predefined list of macros (`MBED_APP_SIZE`, `MBED_APP_START`, `MBED_BOOT_STACK_SIZE`, etc.) and filter out any macros not on the list (instead of filtering out spaces, brackets, etc.). Alternatively, we could explicitly query those on the list. The same static list could be used for any ld files - no need to look into the ld file we process.

(Just a suggestion - not sure how feasible it is.)

LDong-Arm, 2020-10-27:
Closed by #13767

PR #13767 (2020-10-17): CMake: linker preprocess fix for Gcc Arm

### Summary of changes 

This fixes the preprocessing linker script file, the simple macros are passed to the linker. There is a limitation that we will need to fix still. As we are using generators, I am not certain how to replace the values same way as Ninja does. 

This can fix #13765 but the proper fix will need to take place (it might need more time to get these right, same as Ninja). 

#### Impact of changes
closed 2020-10-27 by LDong-Arm