ARMmbed/mbed-os #13783: minimal-printf doesn't display decimals between -1 to 0 correctly

Linguanghsou, 2020-10-19:
### Description of defect

The output of:
    double d = -0.00139;
    printf("d = %f\r\n",d);
with minimal-printf is:
d = 0.00139
but it should be:
d = -0.00139

with setting "target.printf_lib": "std" the output is as expected.

#### Target(s) affected by this defect ?
NUCLEO_F413ZH

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
    double d = -0.00139;
    printf("d = %f\r\n",d);

helpfulchicken, 2020-11-04:
This issue is also affecting me. I'm having to duplicate print calls for negative and positive cases to deal with this.

**Target(s) affected by this defect ?**
NRF52_DK

**Toolchain(s) (name and version) displaying this defect ?**
gcc-arm-none-eabi-9-2019-q4-major

**What version of Mbed-os are you using (tag or sha) ?**
mbed-os-6.4.0  (#ad40b1b2672d)

**What version(s) of tools are you using. List all that apply (E.g. mbed-cli)**
mbed-cli 1.10.4

**Additional details**
I looked into the source code, and I found where this is happening.
In the function "mbed_minimal_formatted_string_double", the integer part of the floating point value is take using this line:
https://github.com/ARMmbed/mbed-os/blob/cdc2d45e05622f375d222b1920a9df148c5acbab/platform/source/minimal-printf/mbed_printf_implementation.c#L265
This works fine as long as negative values are less than or equal to -1.0. 
Otherwise, for values between 0.0 and -1.0, the integer component is "-0", which loses the sign when being saved in a signed integer format. 
Again, for the sake of clarity, writing a floating point value to a signed integer results in loss of sign information if the value in question is between 0.0 and 1.0.

0xc0170, 2020-11-04:
Thanks @helpfulchicken . I commented in the https://github.com/ARMmbed/mbed-os/pull/13784 (fix for this), if we should consider your fix.

PR #13548 (2020-12-10): Add width modifier and prepending zeros for hexadecimal output and decimal precision for floating point

### Summary of changes 

Added width modifier and prepend zeros if the width modifier is given feature to printing integer, hexadecimal and floating point values. So e.g. `printf("%04X", i);` works now the same as the std version.

Added precision specifier for printing floating point values, as it already existed for strings. If no precision specifier is given `MBED_CONF_PLATFORM_MINIMAL_PRINTF_SET_FLOATING_POINT_MAX_DECIMALS` will be used.

This PR also fixes #13783.

#### Impact of chang [... cut]
closed 2020-12-10 by 0xc0170