ARMmbed/mbed-os #13789: CY8CKIT_062_WIFI_BT - mbed-bootloader does not compile

joshheo, 2020-10-20:
### Description of defect

#### Target(s) affected by this defect ?
Cypress - CY8KITC_062_WIFI_BT(CY8C6247BZI_D54)

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

Mbed os 5.12.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed - cli

#### How is this defect reproduced ? 

Compile [ 97.8%]: arm_uc_pal_flashiap_implementation.c
[Error] mbed_config.h@106,70: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@133,69: 'MBED_ROM_SIZE' undeclared (first use in this function); did you mean 'MBED_ROM_START'?
[Error] mbed_config.h@103,70: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@36,69: 'APPLICATION_SIZE' undeclared (first use in this function); did you mean 'ARM_UC_ROT_SIZE'?
[ERROR] In file included from <command-line>:0:0:
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c: In function 'arm_uc_pal_flashiap_get_slot_addr_size':
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:106:70: error: 'MBED_ROM_START' undeclared (first use in this function)
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                     (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                      ^
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:106:70: note: each undeclared identifier is reported only once for each function it appears in
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                     (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                      ^
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:133:69: error: 'MBED_ROM_SIZE' undeclared (first use in this function); did you mean 'MBED_ROM_START'?
 #define MBED_BOOTLOADER_FLASH_BANK_SIZE                             MBED_ROM_SIZE/2                                                                                          // defined by application
                                                                     ^
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:106:87: note: in expansion of macro 'MBED_BOOTLOADER_FLASH_BANK_SIZE'
 #define MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS                     (MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE + KVSTORE_SIZE)                                        // set by application[*]
                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c:106:39: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS'
                                       MBED_CONF_UPDATE_CLIENT_STORAGE_ADDRESS, 0);
                                       ^
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c: In function 'ARM_UC_PAL_FlashIAP_GetActiveDetails':
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:103:70: error: 'MBED_ROM_START' undeclared (first use in this function)
 #define MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS                 (MBED_ROM_START + MBED_BOOTLOADER_SIZE)                                                                  // set by application[*]
                                                                      ^
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c:508:13: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS'
         if (MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS) {
             ^
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:36:69: error: 'APPLICATION_SIZE' undeclared (first use in this function); did you mean 'ARM_UC_ROT_SIZE'?
 #define MBED_BOOTLOADER_SIZE                                        APPLICATION_SIZE                                                                                         // set by application[*]
                                                                     ^
./.\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY\mbed_config.h:103:87: note: in expansion of macro 'MBED_BOOTLOADER_SIZE'
 #define MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS                 (MBED_ROM_START + MBED_BOOTLOADER_SIZE)                                                                  // set by application[*]
                                                                                       ^~~~~~~~~~~~~~~~~~~~
.\modules\storage\pal-flashiap\source\arm_uc_pal_flashiap_implementation.c:508:13: note: in expansion of macro 'MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS'
         if (MBED_CONF_UPDATE_CLIENT_APPLICATION_DETAILS) {
             ^

[mbed] Working path "C:\Users\hujun\mbed-bootloader\mbed-bootloader-4.0.2_0928" (program)
[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\hujun\mbed-bootloader\mbed-bootloader-4.0.2_0928"
       Command: "C:\Python27\python.exe -u C:\Users\hujun\mbed-bootloader\mbed-bootloader-4.0.2_0928\mbed-os\tools\make.py -t GCC_ARM -m CY8CKIT_062_WIFI_BT --profile tiny.json --source . --build .\BUILD\CY8CKIT_062_WIFI_BT\GCC_ARM-TINY -c --app-config configs/internal_flash_no_rot.json"
       Tip: You could retry the last command with "-v" flag for verbose output

I am trying to make mbed os bootloader, but an error occurs during compilation as shown above.

Why does such an error occur??

Help

0xc0170, 2020-10-26:
cc @ARMmbed/team-cypress

0xc0170, 2020-10-26:
> Compile [ 97.8%]: arm_uc_pal_flashiap_implementation.c
[Error] mbed_config.h@106,70: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@133,69: 'MBED_ROM_SIZE' undeclared (first use in this function); did you mean 'MBED_ROM_START'?
[Error] mbed_config.h@103,70: 'MBED_ROM_START' undeclared (first use in this function)
[Error] mbed_config.h@36,69: 'APPLICATION_SIZE' undeclared (first use in this function); did you mean 'ARM_UC_ROT_SIZE'?

Do you follow any tutorial or documentation ? Did you define the above macros in your config?

0xc0170, 2020-11-12:
Please create an issue to bootloader repository. I suspect the target is not supported by default (it might need additional work).

I did not find this target in the configs. Neither where the above config should come.

I'll close this.

closed 2020-11-12 by 0xc0170