ARMmbed/mbed-os #13790: RTC-reset bug on the EFM32GG11_STK3701 boards rev. b06 and b07

petroborys, 2020-10-20:
### Description of defect

Mbed OS *hal-tests-tests-mbed_hal-rtc_reset* test not passed on Silicon Labs STK3701A boards rev. B06 and B07. But the same test passed on the same STK3701A board rev. B01. The test was running wich Mbed OS 6.0 and 6.3 with GCC_ARM 9 2019-q4-major and GCC_ARM 9-2020-q2-update.

Looks like after NVIC_SystemReset() two registers are also reset on STK3701A boards rev. B06 and B07:
```
RTCC->RET[0].REG
RTCC->CNT
```
But on STK3701A boards rev. B01 the current values are saved.

In the attached files hal-tests-tests-mbed_hal-rtc_reset's logs. 

#### Target(s) affected by this defect ?

EFM32GG11_STK3701

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0 and mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

```
mbed test --compile -t GCC_ARM -m EFM32GG11_STK3701 -DMBED_EXTENDED_TESTS=1 -n hal-tests-tests-mbed_hal-rtc_reset
mbed test --run -t GCC_ARM -m EFM32GG11_STK3701 -DMBED_EXTENDED_TESTS=1 -n hal-tests-tests-mbed_hal-rtc_reset -v
```

[report_rtc_reset_rev_b01_SN440109372.txt](https://github.com/ARMmbed/mbed-os/files/5408068/report_rtc_reset_rev_b01_SN440109372.txt)
[report_rtc_reset_rev_b06_SN440145056.txt](https://github.com/ARMmbed/mbed-os/files/5408069/report_rtc_reset_rev_b06_SN440145056.txt)
[report_rtc_reset_rev_b06_SN440145061.txt](https://github.com/ARMmbed/mbed-os/files/5408070/report_rtc_reset_rev_b06_SN440145061.txt)
[report_rtc_reset_rev_b07_SN440174200.txt](https://github.com/ARMmbed/mbed-os/files/5408071/report_rtc_reset_rev_b07_SN440174200.txt)

petroborys, 2020-11-06:
One way to solve the issue is to comment on two rows:

`//RTCC->RET[0].REG = 0;`  - rtc_init(void) function ([targets/TARGET_Silicon_Labs/TARGET_EFM32/rtcc.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Silicon_Labs/TARGET_EFM32/rtcc.c))

`//RTCC->CNT     = _RTCC_CNT_RESETVALUE;`  - RTCC_Reset(void) function ([targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_rtcc.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_rtcc.c))

It looks like a workaround... Also, the [EMLIB (RTCC_Reset() function)](https://docs.silabs.com/mcu/latest/efr32fg1/group-RTCC) is not Mbed's code.

Are there any updates on this issue? Thanks!

0xc0170, 2020-11-12:
cc @ARMmbed/team-silabs

petroborys, 2020-11-12:
Looks like `RMU_ResetControl` in the latest EFM32GG11_STK3701 works incorrectly. My workaround at the moment: 
```
void rtc_init(void)
{
    if (BUS_RegBitRead((&RTCC->CTRL), _RTCC_CTRL_ENABLE_SHIFT) != 0) {
        /* RTCC already up & running */
        return;
    }

    /* Make sure RTCC doesn't get reset (mbed specification) */
    RMU_ResetControl(rmuResetWdog, rmuResetModeLimited);
    RMU_ResetControl(rmuResetCoreLockup, rmuResetModeLimited);
    RMU_ResetControl(rmuResetSys, rmuResetModeLimited);
    RMU_ResetControl(rmuResetPin, rmuResetModeFull);

    /* Set up the RTCC and let it run, Forrest, run */

    //Save time if it has been set.
    time_t t = 0;
    if (RTCC->RET[1].REG == 0) t = rtc_read();

    RTCC_Reset();
    RTCC_Init_TypeDef rtcc_init = RTCC_INIT_DEFAULT;
    rtcc_init.presc = rtccCntPresc_32768;
    RTCC_Init(&rtcc_init);
    RTCC_Enable(true);

    if (RTCC->RET[1].REG == 0) rtc_write(t);
    else RTCC->RET[0].REG = 0;
}

void rtc_write(time_t t)
{
    core_util_critical_section_enter();
    RTCC->RET[0].REG = t - RTCC_CounterGet();
    RTCC->RET[1].REG = 0;
    core_util_critical_section_exit();
}
```

stevew817, 2020-11-13:
Hi @petroborys , would you mind logging a PR with the workaround that works on your end, so that I can better investigate the issue at hand?

closed 2020-12-09 by 0xc0170