ARMmbed/mbed-os #13792: NUCLEO_F429ZI hangs during flash erase

moshe-shahar, 2020-10-20:
### Description of defect

NUCLEO_F429ZI hangs during flash erase operation if `NetworkInterface` is used and connected.
Seems that there is some collision between the flash erase operation the the `NetworkInterface`

#### Target(s) affected by this defect ?
NUCLEO_F429ZI 

#### Toolchain(s) (name and version) displaying this defect ?
GC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
NA

#### How is this defect reproduced? 

1. `mbed import mbed-os-example-blinky`
1. Download [erase-flash-test.patch](https://gist.github.com/moshe-shahar/24ec75d6cc986dcde018a27360572558#file-erase-flash-test-patch) and apply git patch
1. `mbed compile -m NUCLEO_F429ZI -t GC_ARM`
1. Flash and run the application

After a while, the application hangs. If not, reset the target and try again.

yogpan01, 2020-10-20:
@ARMmbed/mbed-os-maintainers who can do triaging for this defect? It is blocking PDMC to migrate to MbedOS 6 version.

0xc0170, 2020-10-20:
cc @ARMmbed/team-st-mcd @ARMmbed/mbed-os-core

0xc0170, 2020-10-20:
@ARMmbed/team-st-mcd Have you seen this, how can emac implementation affects flashing?

jeromecoutant, 2020-10-21:
Are you sure that you don't erase FLASH page where application is ?

moshe-shahar, 2020-10-21:
> Are you sure that you don't erase FLASH page where application is ?

Yes. the application is very small even with _debug_ profile.
```
Name             Origin             Length             Attributes
FLASH            0x0000000008000000 0x0000000000200000 xr
```
and the erase test start from `0x8120000`.

Also, it was tested on more than three HW. So, it's not HW issue.

jeromecoutant, 2020-10-21:
Could you try to add
````
    core_util_critical_section_enter();
````
here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/flash_api.c#L76

and
````
    core_util_critical_section_exit();
````
at the end of the function ?

kjbracey-arm, 2020-10-21:
It could be that the flashing is generating a huge interrupt latency spike that breaks the Ethernet driver's logic - too much for its interrupt handler to process when it gets CPU back.

moshe-shahar, 2020-10-21:
> Could you try to add
> 
> ```
>     core_util_critical_section_enter();
> ```
> 
> here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/flash_api.c#L76
> 
> and
> 
> ```
>     core_util_critical_section_exit();
> ```
> 
> at the end of the function ?

Yes, adding critical section fix or hide the problem.

PR #13802 (2020-11-11): STM32F4 FLASH API update
### Summary of changes 

Fixes #13792

We need to add critical section in flash_erase_sector function

@ARMmbed/team-st-mcd 
@moshe-shahar

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-11-11 by 0xc0170